mantidproject/mantid #16309: LiveData not working at ORNL

AndreiSavici, 2016-05-19:
I tried to load live data from HYSPEC or SEQUOIA, with a preprocessing step Algorithm - SumSpectra (no parameters), and a post processing step Rebin, with parameters "0,10,30000", both with and without quotes.  
### Expected behavior
Works in 3.5
### Actual behavior
On 3.6 (and nightly) I get `Property "PARAMS" is not set to a valid value: "Enter values for this property".`

If I use a post processing script instead  that contains just `output=Rebin(input,Params="0,10,30000")`, it seems to work.
### Platforms affected
I've tested on RHEL7 (analysis.sns.gov) and on my Ubuntu workstation

mantid-roman, 2016-05-19:
@AndreiSavici #16306 should fix it.

NickDraper, 2016-05-20:
duplicate of #16203

closed 2016-05-20 by NickDraper