mantidproject/mantid #16310: Add curve to new layer in multlayer plot seems gets the scaling wrong on Ubuntu

AntonPiccardoSelg, 2016-05-19:
### Expected behaviour

CurveDialog should allow us to add an existing curve to a new layer. The cloned curve should look exactly the same. Expected image is:

[image]

### Actual behaviour

The cloned curve does not appear on the screen (depending on the data set). When changing the scaling of the axis, one can see that the curve was cloned but with an apparently incorrect scaling.  The curve looks like this:

[image]

### Steps to reproduce the behaviour

1. Load the workspace POLREF00014966 from the Unit tests
1. Ungroup it
1. Plot the first spectrum of the first period workspace
1. Add a new layer
1. Double click on the (2) icon 
1. Transfer the old curve
  * Confirm that the new curve has an incorrect scaling

### Platforms affected

I can only see this on Ubuntu. I checked a Nightly Build from 10/05 and it seems OK there.

NickDraper, 2016-05-19:
Multi layer plot in reality are very rarely used

NickDraper, 2016-06-22:
No plan to do anything on this at the moment - Archived

closed 2016-06-22 by NickDraper