mantidproject/mantid #16388: PlotMD: Does not respect the show all error bars flag

martyngigg, 2016-05-24:
Beta test bug.

### Expected behavior

All error bars should be shown if the appropriate option is checked

### Actual behavior

A plot with `plotMD` does not respect the state. 

### Steps to reproduce the behavior

```python
#Some problems with MD algorithms / workflow

#Create an MD workspace, file backed because my system might not have enough memory:

import os
import numpy as np
#

#Create projections workspace for later cutting
from mantid.api import Projection
# Create an identity projection
proj_id = Projection([1,1,0], [-1,1,0], [0,0,1])
proj_id.setType(0, 'r') 
proj_id.setType(1, 'r')
proj_id.setType(2, 'r') 
ws_id = proj_id.createWorkspace()

#Make MD mega file (like gen_sqw / accumulate_sqw)
psi_arr=np.arange(0.0,92.0,2)
run_arr=range(15052,15098)

input_runs = list()
psi = list()
gs = list()
gl = list()
#note that gl, gs etc are not mandtory inputs. If we do not specify them then they are set to zero.

input_runs = list()
psi = list()
gs = list()
gl = list()
for i in range(len(psi_arr)):
    source_file = 'map'+str(run_arr[i])+'_ei400.nxspe'
    input_runs.append(source_file)
    psi.append(float(psi_arr[i]))
    gl.append(0.0)
    gs.append(0.0)

print(psi)
print(input_runs)

md_ws = CreateMD(input_runs, Emode='Direct', Alatt=[2.87,2.87,2.87], Angdeg=[ 90, 90, 90], u=[1, 0, 0,], v=[0,1,0], Psi=psi, Gl=gl, Gs=gs, EFix=400.0,Filename='test.nxs',FileBackEnd=True)
#Up to this point everything works as we would expect it to.

##Problem with PlotMD errorbars
out_md_1d_nopix = CutMD(md_ws, Projection=ws_id, PBins=([-3,0.05,3],[-1.1,-0.9],[-0.1,0.1],[130,150]), NoPix=True)
plotMD(out_md_1d_nopix,error_bars=True,normalization=2)
#After fettling the axes limits, can see that not all errorbars are plotted. Resizing window up and down shows more / fewer errorbars (old behaviour)
#Checkbox for "show all errorbars" was checked (since on by default)
```

I have the data for this script.

### Platforms affected
All

FedeMPouzols, 2016-05-24:
Aha, if I'm interpreting correctly the differences between `MantidMatrixCurve` and `MantidMDCurve` this seems easy to fix. I think that MD curves have never paid any attention to the MantidPlot setting "draw all error bars", and probably the change of default in v3.7 has brought this up.

martyngigg, 2016-05-24:
@FedeMPouzols That sounds plausible. I'm glad it wasn't to hard to find!

closed 2016-05-27 by NickDraper