mantidproject/mantid #16398: H5Utils issues with long and long long types

quantumsteve, 2016-05-24:
### Expected behavior

H5Utils "does the right thing" when loading or writing either a `std::vector<long>` or `std::vector<long long>` 

### Actual behavior

On certain platforms, the user will encounter a linking error.

### Platforms affected

On Windows `long` has the same length as `int32_t`, but it is a ​*different*​ type.
On Windows `long long` has the same length as `int64_t`, and it is the ​*same*​ type.
http://rextester.com/EEFS42414

On OS X & Linux `long` has the same length as `int64_t`, and is the ​*same*​ type.
On OS X & Linux `long long` has the same length as `int64_t`, but is a ​*different*​ type.
http://cpp.sh/7u2t

quantumsteve, 2016-05-24:
HDF5 has `H5T_NATIVE_LONG`, `H5T_NATIVE_ULONG`, `H5T_NATIVE_LLONG`, `H5T_NATIVE_ULLONG` types, which (assuming the NeXus format allows them) may be useful. Otherwise, we may need our own logic to steer `long` on Win32 to `int32_t` and `long long` on Linux to `int64_t`.

peterfpeterson, 2016-05-25:
The good news is we don't support win32.

quantumsteve, 2016-05-25:
By Win32 I mean 64-bit Windows, which we *do* support. `WIN32` is a macro often used to determine a Windows machine and defined in both 32 and 64-bit builds..

closed 2021-03-03 by stale[bot]