mantidproject/mantid #16425: Plotly support in SavePlot1D

peterfpeterson, 2016-05-25:
Example of using plotly with matplotlib:
```python
import matplotlib.pyplot as plt
import matplotlib.mlab as mlab
import numpy as np
import plotly.offline as pltly

n = 50
x, y, z, s, ew = np.random.rand(5, n)
c, ec = np.random.rand(2, n, 4)
area_scale, width_scale = 500, 5

fig, ax = plt.subplots()
sc = ax.scatter(x, y, c=c,
                s=np.square(s)*area_scale,
                edgecolor=ec,
                linewidth=ew*width_scale)
ax.grid()

if True:  # full html page
    pltly.plot_mpl(fig, filename='/tmp/deleteme')
else:  # just the div
    pltly.plot_mpl(fig, filename='/tmp/deleteme',
                      output_type='div', include_plotlyjs=False)
```

peterfpeterson, 2016-05-25:
@stuartcampbell is interested in this

AndreiSavici, 2016-05-25:
Note: require plotly version>1.9.4

PR #16776 (2016-07-11): SavePlot1D and plot.ly
Add support for exporting [plotly](https://plot.ly/python/) in `SavePlot1D`.

**To test:**

Try saving both images and html. You must have the plotly python api in your path for that functionality to work. `OutputType="plotly-full"` will generate full html pages that you can view.

Fixes #16425.

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [x] Is the code of an acceptable quality?
- [x] Does the code conform to the [coding standards](http://www.mantidpro [... cut]
closed 2016-07-11 by FedeMPouzols