mantidproject/mantid #16426: Script error for ISIS direct inelastic

martyngigg, 2016-05-26:
From Jon Taylor:

I get an issue with Iliad, maybe in the production stuff for isis this is not a problem
 
line 1791 Direct energy conversion.py
 
```
KeyError: 'Property with name: nullify_negative_signal is not among the class properties '
  at line 56 in '/Users/jonathantaylor/Dropbox/SrFeO2/SrFeO2.py'
  caused by line 144 in '/Applications/MantidPlot.app/scripts/Inelastic/Direct/dgreduce.py'
  caused by line 525 in '/Applications/MantidPlot.app/scripts/Inelastic/Direct/DirectEnergyConversion.py'
  caused by line 717 in '/Applications/MantidPlot.app/scripts/Inelastic/Direct/DirectEnergyConversion.py'
  caused by line 1791 in '/Applications/MantidPlot.app/scripts/Inelastic/Direct/DirectEnergyConversion.py'
  caused by line 205 in '/Applications/MantidPlot.app/scripts/Inelastic/Direct/PropertyManager.py'
  caused by line 244 in '/Applications/MantidPlot.app/scripts/Inelastic/Direct/ReductionHelpers.py'
 ```
 
changing line 1791 to:
```
if prop_man.check_background: #and prop_man.nullify_negative_signal:
```

helps.

Tested on OSX

@abuts If this is not a problem for ISIS in production then we can probably drop this back to `v3.8`

abuts, 2016-05-26:
'nullify_negative_signal' property has been introduced  by #14456 on 18th of November in Mantid 3.6.0. 
It has been implemented as autoproperty, automatically generated from an appropriate instrument parameter file, which has such property. 
As such it works with all recent Mantid versions given the IDF is also recent, so I can not reproduce the issue. 

If local Mantid settings done in a way that old IDF is used or (may be, is it possible?) fully overwritten by old IDF stored in nxs file this error would occur. 

I need a bit more information on when this error occurs to be able to reproduce the  issue and think about the ways to deal with it.

martyngigg, 2016-05-26:
@Jon-Taylor Can you provide any additional info for this?

NickDraper, 2016-06-01:
@abuts If we are going to include this in R3.7 then we need to get the PR in today.

mducle, 2016-06-01:
Tatiana also reports the same error... we can have a work-around in Direct_energy_conversion.py to check if this property exists and if not to assume it is false (e.g. default behaviour before November last year).

NickDraper, 2016-06-01:
Thanks @mducle, we will take a look at this, could you give me the steps to reproduce the problem?

mducle, 2016-06-01:
@martyngigg @NickDraper Sorry I was just with Tatiana trying to find out what was wrong.

It seems that an old version of Mantid on her Windows 7 laptop installed all the instrument definition into a user `/AppData/Roaming/MantidInstall/mantid/instrument` folder which is searched first. The newer definitions `c:/MantidInstall/instrument` is searched second (these paths were returned by `mantid.api.ExperimentInfo.GetInstrumentFilename('MARI')` and `mantid.kernel.ConfigService.Instance().getInstrumentDirectories()`). I have no idea where the first path (with the old instrument files) is defined - the preference dialogue only lists the second path (new definitions). I guess it could be a registry setting, but haven't looked.

Anyway, the work-around is:

```
diff --git a/scripts/Inelastic/Direct/DirectEnergyConversion.py b/scripts/Inelastic/Direct/DirectEnergyConversion.py
index f16b196..2adfa83 100644
--- a/scripts/Inelastic/Direct/DirectEnergyConversion.py
+++ b/scripts/Inelastic/Direct/DirectEnergyConversion.py
@@ -1788,7 +1788,7 @@ class DirectEnergyConversion(object):
         if prop_man.energy_bins: # It should already be a distribution.
             ConvertToDistribution(Workspace=result_ws)
         # nullify negarive signals if necessary
-        if prop_man.check_background and prop_man.nullify_negative_signal:
+        if prop_man.check_background and ('nullify_negative_signal' in dir(prop_man) and prop_man.nullify_negative_signal):
             zeroBg = CreateWorkspace(DataX='0,1',DataY=0,DataE=0,UnitX='TOF')
             result_ws=RemoveBackground(result_ws,BkgWorkspace=zeroBg,Emode='Direct',NullifyNegativeValues=True)
             DeleteWorkspace(zeroBg)
```

mducle, 2016-06-01:
To reproduce the error, use the attached MARI_Parameters_Old.xml - everything works with MARI_Parameters_New.xml.

[MariParameters.zip](https://github.com/mantidproject/mantid/files/293255/MariParameters.zip)

martyngigg, 2016-06-01:
@mducle Just so I understand what has happened:

* there was an older version of Mantid on the machine that downloaded various files to `/AppData/...`
* a newer version of Mantid was installed but the files remain in `/AppData/...`
* the `_Parameters` file in `/AppData` now takes precedence regardless of the newer version that is in the package.

Is that about right?

In any case I will apply your fix but it sounds like we have uncovered and unintended side effect of instrument download mechanism.

mducle, 2016-06-01:
@martyngigg Yes. That's about right. The `/AppData/...` folder only has the instrument files - I guess you mean that it's put there if you enable `UpdateInstrumentDefinitions.OnStartup` in `Mantid.properties`?

martyngigg, 2016-06-01:
@mducle I think it's actually put there regardless of the setting of `UpdateInstrumentDefinitions.OnStartup`. The parameter just controls whether the download happens or not.

@NickDraper I think we will need to have a think what we can do to avoid this confusion in future versions.

NickDraper, 2016-06-01:
Issue #16506 created to prevent this happening in the future

abuts, 2016-06-01:
I am on AL now but its an important issue so I am trying to look into this. Still can not understand how it can happens, as Mantid pulls new IDF-s into /Appdata and uses it so both IDF-s should be correct. 
Communicating with Tatiana about this. 
She says that she have deleted old IDF from Appdata. 

As I understand,  @mducle fix is applied so it should be OK, but the issue is certainly deeper.

mducle, 2016-06-01:
> Mantid pulls new IDF-s into /Appdata and uses it so both IDF-s should be correct. 

Does this only occur if `UpdateInstrumentDefinitions` setting is set to true? Maybe what happens is that when a new installation occurs this parameter is set to false, so the instrument definition is not updated. Is there a way to set this in MantidPlot? Or do you have to change `Mantid.properties`?

martyngigg, 2016-06-01:
@mducle Yes, the downloading only occurs `UpdateInstrumentDefinitions` is set to `true`. It should always be switched on so I'm not 100% sure why they didn't download. The only think we could come up with is that it is trying to download them but is failing for some reason...

NickDraper, 2016-06-01:
Alex,

Thanks for responding.

Martyn is putting Duc’s fix in, and I’m looking at adding a ClearCaches algorithm for 3.8 that should be able to get people out of this if it every happens again.

Regards,
Nick Draper

mducle, 2016-06-01:
@martyngigg 

> It should always be switched on so I'm not 100% sure why they didn't download.

On my version of `Mantid.properties`, I have:

```
# Whether to check for updated instrument definitions on startup of Mantid
UpdateInstrumentDefinitions.OnStartup = 0
UpdateInstrumentDefinitions.URL = https://api.github.com/repos/mantidproject/mantid/contents/instrument

# Whether to check for newer mantid versions on startup
CheckMantidVersion.OnStartup = 0
CheckMantidVersion.GitHubReleaseURL = https://api.github.com/repos/mantidproject/mantid/releases/latest
CheckMantidVersion.DownloadURL = http://download.mantidproject.org
```

Tatiana also has the same...

martyngigg, 2016-06-01:
@mducle Is that a downloaded nightly?

mducle, 2016-06-01:
Yes. 

(Actually it was a downloaded 3.6.0 with self-compiled binaries installed on top, but I've just now checked with a fresh nightly download and it's the same).

martyngigg, 2016-06-01:
Thanks, @mducle. It seems that there is indeed a bug in generating the properties file that has the downloading turned off when it should be on. I'm looking into it now.

PR #16510 (2016-06-02): DirectEnergyConversion.py: Check for newer attributes
Adds a simple check for the existence of an attribute before trying to access it.

**To test:**

See issue for data and script.

Fixes #16426

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [x] Is the code of an acceptable quality?
- [x] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)? Is it well structured with small focussed classes/m [... cut]
closed 2016-06-02 by NickDraper