mantidproject/mantid #16428: Sample transmission calculator: Single bin not handled well

martyngigg, 2016-05-26:
Beta-test bug from Jon Taylor.

### Expected behavior

Error dialog should be shown

### Actual behavior

Terminate/continue is displayed

### Steps to reproduce the behavior

* Open `Interfaces->General->Sample Transmission Calculator`
* Set Low=0.1,Width=0.1,High=0.2
* Set `ChemicalFormula=V`
* Hit Calculate.

### Platforms affected

All

martyngigg, 2016-05-26:
@raquelalvarezbanos Do you have time to look at this for the release?

raquelalvarezbanos, 2016-05-26:
@martyngigg Yes, I can have a look at it. Looks like an easy bug to fix, if that's the case I can fix it, if it turns out to be more complicated I may need help (that interface is new to me). I'll have a look now.

martyngigg, 2016-05-26:
Thanks. Unfortunately the original author is not around but hopefully it won't be too complicated.

raquelalvarezbanos, 2016-05-26:
Done, it was not too difficult. @martyngigg feel free to assign me other issues I can help with, I am working on maintenance tasks at the moment.

martyngigg, 2016-05-26:
Thanks, @raquelalvarezbanos. I'll keep it in mind.

PR #16434 (2016-05-26): Fix bug in SampleTransmission Calculator
Fixes an unhandled exception thrown when the output workspace has less than two X values (`PreviewPlot` is not able to handle that case).

**To test:**

- Open Interfaces->General->Sample Transmission Calculator
- Set Low=0.1,Width=0.1,High=0.2
- Set ChemicalFormula=V
- Hit Calculate.
- An error message should be shown indicating the problem.

Fixes #16428

Does not need to be in the release notes, issue found during beta testing period.

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual [... cut]
closed 2016-05-31 by NickDraper