mantidproject/mantid #16429: Sequential Fitting gives bad lexical cast

martyngigg, 2016-05-26:
Beta-test bug reported by Jon Taylor:

"Sequential fitting on multiple spectra in a workspace produces
Error in execution of algorithm PlotPeakByLogValue:
bad lexical cast: source type value could not be interpreted as target"

"In the online help the range parameter is not defined"

You may need to get in touch with Jon for more information.

martyngigg, 2016-05-26:
@mantid-roman Do you have time to look at this and #16389?

mantid-roman, 2016-05-26:
Yes.

AntonPiccardoSelg, 2016-06-01:
I tried to replicate the  issue but it doesn't seem to be appearing for me. What I did:
1. Select Sequential Fitting
2. Add the same workspace several times and increment the spectrum for each of them
3. Select "Plot against log" and select a log to plot against
4. Run Fit

All seems fine. 

Initially I thought that the Range column could be used to define the spectrum range, but this is not true. This selects the bin values. If we set a value of the incorrect format, e.g. 2-5 then I get the reported bad lexical cast error. If the correct format is used, i.e. 2:5 then all is fine. I agree that we need better documentation, e.g. tooltip information for the Range column. 

I figure that the issue is related to a misuse the Range column. To prevent this, we should add documnetation here. I have contacted Jon to check what his exact steps were.

PR #16507 (2016-06-02): Improve exception information when range is misused in sequential fit UI
Added tooltip informationf or the range column in the sequential fit UI. It was not fully clear what to insert, and what the insertion format was (colon-separated range)

**To test:**

1. Load the file GEM38370_Focussed.nxs
2. Plot spectrum 2
3. Open the fit browser on top of the plot
4. Add a linear background fit function 
5. Select Sequential Fit --> the sequential fit browser should open
  * Confirm that there is tooltip for the `Range column`
6. Enable `Plot against log` and set the log vlaue to `Period log`
7. In th [... cut]
closed 2016-06-02 by NickDraper