mantidproject/mantid #16430: Multi-dataset fitting: misc beta bugs

martyngigg, 2016-05-26:
Beta-test bug from Jon Taylor.

"Uses the term workspace index in the UI not spectrum  number but has all spectra checkbox
 fitting will allow a tile plot for any number if spectra to be generated (export all plots). which crashes mantid for say 285 spectra
initiating the fit makes fit algorithm unresponsive with 285 spectra and simultaneous fit selected
editing fit parameters shrinks text edit window in a way that makes it impossible to read

"Plot guess also plots the fit in the ui when the algorithm completes"

You may need to contact Jon Taylor for further information.

martyngigg, 2016-05-26:
@shahrozahmed1 Are you available to take a look at these issues for the release?

shahrozahmed1, 2016-05-26:
yes @martyngigg, I'll take a look at this asap

martyngigg, 2016-05-26:
Thanks. Feel free to grab Roman to point you in the right direction.

closed 2016-05-31 by NickDraper