mantidproject/mantid #16454: Refactor MatrixWorkspace::getSpectrum

SimonHeybrock, 2016-05-27:
To clean up existing code and reduce the amount of code duplication, the `MatrixWorkspace` interface could be refactored as followed:

1. `getSpectrum()` returns a reference instead of a pointer.

2. The return value of `getSpectrum()` should be covariant, i.e., we will have

  ```cpp
  ISpectrum &MatrixWorkspace::getSpectrum(size_t);
  Histogram1D &Workspace2D::getSpectrum(size_t);
  EventList &EventWorkspace::getSpectrum(size_t);
  ```

3. `EventWorkspace::getSpectrum()` now returns `EventList &` so we can remove `EventWorkspace::getEventList` and `EventWorkspace::getEventListPtr`.

The following is an example of duplicate code that could be eliminated:

```cpp
void GroupDetectors2::moveOthers(API::MatrixWorkspace_const_sptr inputWS, API::MatrixWorkspace_sptr outputWS) {
    // ...

    // The input spectrum we'll copy
    const ISpectrum *inputSpec = inputWS->getSpectrum(sourceIndex);

    // Destination of the copying
    ISpectrum *outputSpec = outputWS->getSpectrum(outIndex);

    // Copy the data
    outputSpec->dataX() = inputSpec->dataX();
    outputSpec->dataY() = inputSpec->dataY();
    outputSpec->dataE() = inputSpec->dataE();

    // Spectrum numbers etc.
    outputSpec->setSpectrumNo(inputSpec->getSpectrumNo());
    outputSpec->clearDetectorIDs();
    outputSpec->addDetectorIDs(inputSpec->getDetectorIDs());

    // ...
}

void GroupDetectors2::moveOthersEvent(DataObjects::EventWorkspace_const_sptr inputWS, DataObjects::EventWorkspace_sptr outputWS) {
    // ...

    // The input spectrum we'll copy
    const EventList &inputSpec = inputWS->getEventList(sourceIndex);

    // Destination of the copying
    EventList &outputSpec = outputWS->getEventList(outIndex);

    // Copy the data
    outputSpec += inputSpec;

    // Spectrum numbers etc.
    outputSpec.setSpectrumNo(inputSpec.getSpectrumNo());
    outputSpec.clearDetectorIDs();
    outputSpec.addDetectorIDs(inputSpec.getDetectorIDs());

    // ...
}

```
After the suggested changes, this could be reduced to:

```cpp
template<class TIn, class TOut>
void GroupDetectors2::moveOthers(const TIn &inputWS, TOut &outputWS) {
    // ...
    inputWS->getSpectrum(sourceIndex) = outputWS->getSpectrum(outIndex);
    // ...
}
```
(Note that also allows `TIn` = event and `TOut` = ws2d if we provide `Histogram1D &Histogram1D::operator=(const ISpectrum &)`.

martyngigg, 2016-05-27:
I think this is a great idea. Reducing this type of copy-and-paste is only a good thing in my opinion

:+1:

martyngigg, 2016-05-27:
The Python layer should be able to handle the covariant return types but that is the only place I can see some complication might arise. I think it will all be solvable though.

FedeMPouzols, 2016-05-27:
:+1:  but if `getEventList()` and friends are removed, could that break some user scripts?
We don't use it much from Python in the Mantid code itself.

SimonHeybrock, 2016-05-27:
@FedeMPouzols For now, we can let the python `getEventList` forward to `getSpectrum` to provide a legacy interface.

martyngigg, 2016-05-27:
We can also add a python deprecation warning so that when it is used users are warned that it will be removed

SimonHeybrock, 2016-05-27:
@martyngigg Do we have a standard way of adding such deprecation warnings?

martyngigg, 2016-05-27:
@SimonHeybrock You just have to use the standard Python warning mechanism, for example [Unit.cpp]( https://github.com/mantidproject/mantid/blob/d7de602d44c966e2846ab91e9c236136ad9a9654/Framework/PythonInterface/mantid/kernel/src/Exports/Unit.cpp#L17)

martyngigg, 2016-05-27:
Although thinking about it that could be made into a `boost::python::return_value_policy` and then it wouldn't need the proxy function. That never occurred to me before!

PR #16463 (2016-06-14): Refactor getSpectrum to return a reference with covariant return type
Make sure to merge https://github.com/mantidproject/mantid/pull/16477 before this.

Fixes #16454.

See issue for description.

**To test:**

Code review:
Mostly the changes are simple. One thing to pay special attention to:
```cpp
auto spec = ws->getSpectrum(i);
```
Previously `auto` would be a pointer, now it will make a **copy** of the spectrum. Make sure I changed all (relevant) occurences to:
```cpp
auto &spec = ws->getSpectrum(i);
```

Internal change, no release notes.

**Important:**

**This commit changes the interfa [... cut]
closed 2016-06-14 by martyngigg