mantidproject/mantid #16455: Changes to ReflectometryReductionOne (Auto) requested by scientists

raquelalvarezbanos, 2016-05-27:
In a meeting with Max and Jos yesterday we looked at `ReflectometryReductionOne` in detail. We need to make some changes to the algorithm:

- In the workflow diagram, there is a `Scale` box at the top, it should not be there but at the end right after the question "Scale factor given?". Aso, in the diagram it seems that if we `Rebin` in `Q` we won't apply the scaling, which is not true: the scaling is applied as long as a factor is given, independently of the `Rebin` in `Q`.
- They don't want to have a `Rebin` in wavelength, `CropWorkspace` is fine but not the rebin, which should be removed. As it is going to be removed, the input property `WavelengthStep` is no longer needed, we should remove it too.
- `ProcessingInstructions` need to be added to the flow chart and documented better.
- The monitor should be optional. This was also requested by Tim and has been addressed in PR https://github.com/mantidproject/mantid/pull/15500
- We need an additional input property to apply flood corrections. This property was suggested to be called `DetectorEfficiencyCorrection`. This step should replace the two boxes `Multidetector?` and `Perform Direct Beam Normalization`. This will be a workspace which can contain:
 - A single data point, in this case this value should be used as a normalization constant shared across the different spectra in the input ws.
 - A list of spectra with a single value: in this case each spectrum will have its own normalization constant.
- `CreateTransmissionWorkspaceAuto` needs to be discussed at some point with them.
- Currently, if no transmission run is given no corrections are applied by default. Scientists would like the algorithm to read the type of corrections from the parameter file (see INTER for an example, where `polynomial` is the defined correction) and use it as a default. If no correction is found in this file, a warning message should be printed.
- `CorrectDetectorPositions` should be defaulted to `False`. This has caused some confusion and it seems that in most of the cases they don't want to correct the detectors.
- The workflow diagram does not document the calculations performed before outputting the workspace in wavelength. Specifically they pointed out that background subtraction and polarization corrections are missing.
- We need to document the difference between processing instructions specified with "-" and ",". Do they produce different results? We should add a section explaining this.

The workflow diagram is getting really long, we may need to consider splitting it into different sections.

raquelalvarezbanos, 2016-05-27:
@MattKing06 Please have a look at this issue and split into different tasks. Let me know if you have any doubts.

MattKing06, 2016-06-27:
"The workflow diagram does not document the calculations performed before outputting the workspace in wavelength. Specifically they pointed out that background subtraction and polarization corrections are missing."

We will need to discuss this issue as the polarization corrections are only performed in ReflectometryReductionOneAuto and hence why the steps are not involved in the ReflectometryReductionOne workflow diagram. After discussing this with @raquelalvarezbanos, we were wondering if there was any reason why we can't perform polariziation corrections in ReflectometryReductionOne? @OwenArnold if you could weigh in on this we would really appreciate it

OwenArnold, 2016-06-27:
I think Harry implemented this first time around. `ReflectometryReductionOneAuto` is different from `ReflectometryReductionOne` only in that it drags out parameters from the Instrument Parameter file, and thus makes many inputs optional. There is therefore no theoretical reason why this couldn't/shouldn't be put into `ReflectometryReductionOne`. Are users asking for this?

MattKing06, 2016-06-27:
They haven't explicitly ask for this to be a part of `ReflectometryReductionOne`, however for continuity, we were thinking of having the option to set `PolarizationCorrection` properties manually in `ReflectometryReductionOne` as we do with other properties that are shared between `ReflectometryReductionOne` and `ReflectometryReductionOneAuto`. Hopefully, we can gain some consistency between the two algorithms as at the moment, it looks like the `Auto` version is doing a few more things that we don't get in `ReflectometryReductionOne`

OwenArnold, 2016-06-27:
I think the consistency is a good argument. I think the current implementation is a little messy and not well enough documented, so I'd like to see that cleared up while the code is moved to `ReflectometryReductionOne`. Any assumptions about input workspaces being the ADS should be refactored out too.

MattKing06, 2016-06-27:
I will open up a new issue for adding `PolarizationCorrection` into `ReflectometryReductionOne` and then hopefully we can document this step properly into the Workflow diagram.

OwenArnold, 2016-06-27:
@raquelalvarezbanos. From what you said in the description
```
Currently, if no transmission run is given no corrections are applied by default. Scientists would like the algorithm to read the type of corrections from the parameter file (see INTER for an example, where polynomial is the defined correction) and use it as a default. If no correction is found in this file, a warning message should be printed. #16518
```
I just wanted to be sure that you know that `CorrectionAlgorithm` is all the users need to select in order to get that Transmission correction mode selected.

raquelalvarezbanos, 2016-06-27:
@OwenArnold Yes, my understanding was that `CorrectionAlgorithm` is used to select the type of corrections (either polynomial or exponential) and then, for instance in the case of INTER, parameter `polystring` is used to populate the polynomial coefficients. 

However, I think @MattKing06 (Matt, correct me if I am wrong) looked into this and this is the default behaviour in `ReflectometryReductionOneAuto`, that is, in `ReflectometryReductionOneAuto`, the default value for `CorrectionAlgorithm` is `AutoDetect`, which searches for the type of corrections in the IDF and populate the coefficients. I think there is some confusion with the two algorithms, `RRO` and `RROA`, scientists usually refer to `ReflectometryReductionOne`'s documentation (probably because the workflow diagram there is more detailed) but use `ReflectometryReductionOneAuto` (for instance via the interface).

I think there is no need to fix this, as the functionality is already in `RROA` (besides, isn't this algorithm the one that is supposed to search in the IDF and use the information there to populate `RRO`'s properties?). Perhaps, what we should do to make things more clear is add a section somewhere (maybe in `RRO` documentation) explaining how both algorithms relate. If you both agree that this is something that should not be fixed, @MattKing06 could you cross out the corresponding item in the list and open a new one to document this better? As you are leaving this week, please assign it to me.

PranavBahuguna, 2016-08-09:
@raquelalvarezbanos Could you provide clarification on task 10? As far as I'm aware, the output workspaces already provide the results in both IvsLam and IvsQ form.

raquelalvarezbanos, 2016-08-10:
@PranavBahuguna: that is something we first have to discuss with scientists

gemmaguest, 2018-03-21:
I'm closing this issue as it is old and out of date and all of the main points have been addressed

closed 2018-03-21 by gemmaguest