mantidproject/mantid #16456: New algorithm for background subtraction

raquelalvarezbanos, 2016-05-27:
This needs to be discussed with scientists. We need a new algorithm that processes background instructions for background subtraction. This is applicable to both linear detectors and 2D detectors.

Blocked by:

- [ ] https://github.com/mantidproject/mantid/issues/16455

gemmaguest, 2018-04-05:
This will include:
- [ ] Instrument background
  - the offspec reduction scripts have a simple background correction
  - SNS also have this?
- [ ] Offspecular background
  - ILL working on this?
- [ ] Dark counts
  - Rob at ISIS working on this?
  - SANS also interested

gemmaguest, 2019-01-15:
Notes on SNS reflectometry background subtraction:

- They use the average of the 3 pixels each side of the peak. If they were to start again, they would fit the whole curve rather than use the 3 points each side, which already have a big error. This could also work in 2D by fitting the surface.
- They do a second correction for the wide shoulder outside the slits. This depends on the flux. This is a tiny correction. It's not always obvious where the edges are - it's somewhat guesswork.
- Dark counts (electric noise) are even tinier. SANS do it but it's not significant for reflectometry.

soininen, 2019-01-16:
At ILL we use [`CalcualtePolynomialBackground`](http://docs.mantidproject.org/nightly/algorithms/CalculatePolynomialBackground-v1.html) with 0, 1 or 2 degree polynomial to do the background subtraction. The workspace needs to be `Transpose`d before fitting the polynomials. This is all done in `ReflectometryILLPreprocess`. The users have the option to select the degree and a pixel range to exclude from the fit.

gemmaguest, 2019-03-18:
I'm re-opening this because there are more types of background calculation we would like to add

PR #24825 (2019-03-15): Basic algorithm for background subtraction
**Description of work.**

A new algorithm for reflectometry background subtraction. It has 2 different methods. The first is based on a script used occasionally at offspec which calculates the background by grouping the detectors which contain the background and dividing by the number of detectors in the background range. Then deducting this from the original input. This is written in AverageSpectrumBackground The second method uses an existing algorithm called calculatePolynomialBackground.  Both methods are called by ReflectometryBackgroundSubtracti [... cut]
PR #25540 (2019-06-11): Add new method for reflectometry background subtraction
**Description of work.**
This PR adds an additional method for background subtraction to the algorithm `ReflectometryBackgroundSubtraction`. This is the method used by SNS and uses their algorithm `LRSubtractAverageBackground`.

**To test:**
The test the output of `LRSubtractAverageBackground` and `ReflectometryBackgroundSubtraction` equal using this script
```
#create a sample workspace
dataX = [1, 2, 3]
background = [2, 2, 2, 2, 2, 2]
peak = [5, 5, 5, 5, 5, 5]
dataY = background + peak + background 
ws = CreateWorkspace(dataX, dataY, NS [... cut]
closed 2019-06-11 by martyngigg