mantidproject/mantid #16466: Differences in SofQWNormalisedPolygons between v3.6.1 and current

mducle, 2016-05-27:
### Expected behavior

Output of SofQW3 in version 3.6.1 is the same as at present.

### Actual behavior

A slight difference in the generated S(q,w) map as shown by the difference map:
[image]

### Steps to reproduce the behavior

Attached is a zip of the necessary nxs files:
[sofqw3_differences_nxs.zip](https://github.com/mantidproject/mantid/files/286473/sofqw3_differences_nxs.zip)

Use this script:

```
ws1 = Load('/babylon5/Public/duc/mt361/mt361/MgAgSb_300K_Al')
ws2 = Load('/babylon5/Public/duc/mt362/mt362/MgAgSb_300K_Al')
wsd = ws2 - ws1
y = wsd.extractY()
id = np.logical_not(np.isnan(y))
print "Difference in reduced data = "+str(np.sum(np.sum(y[id])))

ws1s = Load('/babylon5/Public/duc/mt361/mt361/MgAgSb_300K_A
[sofqw3_differences_nxs.zip](https://github.com/mantidproject/mantid/files/286471/sofqw3_differences_nxs.zip)
l_SQW')
ws2s = Load('/babylon5/Public/duc/mt362/mt362/MgAgSb_300K_Al_SQW')
wsd = ws2 - ws1
y = wsd.extractY()
id = np.logical_not(np.isnan(y))
print "Difference in S(q,w) maps = "+str(np.sum(np.sum(y[id])))

ws_SQW = SofQWNormalisedPolygon(ws1,[0,0.1,9],'Direct',50); 
ws_SQW = Transpose('ws_SQW')

wsd = ws_SQW - ws1
y = wsd.extractY()
id = np.logical_not(np.isnan(y))
print "Difference between loaded S(q,w) map constructed by v3.6.1 and that calculated now = "+str(np.sum(np.sum(y[id])))

wsd = ws_SQW - ws2
y = wsd.extractY()
id = np.logical_not(np.isnan(y))
print "Difference between loaded S(q,w) map constructed by v3.6.20160510 and that calculated now = "+str(np.sum(np.sum(y[id])))
```

Results obtained:

```
Difference in reduced data = 0.0
Difference in S(q,w) maps = 1260.65851468
Difference between loaded S(q,w) map constructed by v3.6.1 and that calculated now = 1260.2388751
Difference between loaded S(q,w) map constructed by v3.6.20160510 and that calculated now = -0.419639583842
```

Expect all differences to be zero or small. Last difference is probably due to rounding errors in the nxs file?

### Platforms affected

Linux (isiscompute - RHEL7) and Windows 7. Not checked on other platforms.

mducle, 2016-05-27:
@martyngigg As we said before, the difference might not be a bug as such - if it's caused by a bug fix affecting the calculation of the polygons as you thought. It would be good to know which version is more 'correct'.

martyngigg, 2016-05-27:
Thanks for the comprehensive report @mducle. I'm taking a look now.

martyngigg, 2016-05-27:
@mducle I have found the change that causes the difference between the current nightly & `3.6.1`. It was  indeed a fix for bug uncovered while trying to use the algorithm for reflectometry.

The issue (#15188) and pull request (#15321) give quite a good overview of what was fixed. Essentially it is this [commit](/mantidproject/mantid/commit/cc13009aa0aa8daa5307997a02ee59a091d4cc01#diff-adbbe4dc246ff84f429da26034cbe698) that fixes a bug in the normalization and thus gives rise to the current differences. The old code was not removing the bin-width normalization correctly and for some reason this had gone unnoticed in the cases we have covered so far.

Is is acceptable to keep the current behaviour? If so I will just add something to the release notes regarding the fix.

mducle, 2016-05-27:
@martyngigg Great. Thanks for looking at this. I think we should keep the current behaviour and add a comment to the release notes.

closed 2016-05-27 by mducle

martyngigg, 2016-05-27:
Thanks.

Release notes updated - https://github.com/mantidproject/mantid/commit/9a1a1bfd5718290a633a1a827b76252bb58443e3