mantidproject/mantid #16471: Table row spacing is wider than it used to be

tom-perkins, 2016-05-27:
The table row spacing is wider in the beta than it used to be in 3.6. This might be inconvenient for users on laptops or with smaller screens.

In addition, the rows have changed from being 0-indexed to being 1-indexed:

*Mantid 3.6.0:*
[image]

*Mantid 3.7 beta:*
[image]

(Reported on Windows, not sure if it's the same on other platforms)

NickDraper, 2016-06-22:
Is this a problem?

tom-perkins, 2016-06-22:
Only if you have a really small screen (I think this was something reported in the beta test). We should make sure that the change from 0-indexing to 1-indexing hasn't broken anything, though.

tom-perkins, 2016-10-14:
This can probably be closed as it is not a problem for anybody

closed 2016-10-14 by NickDraper