mantidproject/mantid #16480: Fit functions without documentation page

FedeMPouzols, 2016-05-31:
### Expected behavior

Every fit function should have a documentation page: http://docs.mantidproject.org/nightly/fitfunctions/index.html

### Actual behavior

There is no documentation for the following functions:

- FullprofPolynomial
- Polynomial
- Bk2BkExpConvPV (but there is documentation for BackToBackExponential)
- ResolutionConvolvedCrossSection
- ElasticDiffRotDiscreteCircle
- ElasticDiffSphere
- FickDiffusion
- InelasticDiffRotDiscreteCircle
- InelasticDiffSphere
- ComptonPeakProfile?, ComptonScatteringCountRate? GaussianComptonProfile? GramCharlierComptonProfile? (there is documentation for MultivariateGaussianComptonProfile)
- MultiDomainFunction
- PawleyParameterFunction
- PeakParameterFunction
- PoldiSpectrumConstantBackground, PoldiSpectrumDomainFunction, PoldiSpectrumLinearBackground, PoldiSpectrumPawleyFunction?
- ReflectivityMulf
- Resolution
- ThermalNeutronBk2BkExpAlpha, ThermalNeutronBk2BkExpBeta, , ThermalNeutronBk2BkExpSigma, ThermalNeutronDtoTOFFunction  (there is documentation for ThermalNeutronBk2BkExpConvPVoigt)
- UserFunctionMD (there is documentation for UserFunction).

(...more may be added later on)

### Steps to reproduce the behavior

You won't find the rst pages for these functions under: https://github.com/mantidproject/mantid/tree/master/docs/source/fitfunctions, so they are not found in the help browser or online pages.

Note that for some of these functions there might be old wiki pages.

For some of these functions (the groups with similar names) it may make sense as there is not much to say, but it would be good to have a documentation page that just states what is the exact difference with respect to the "base" function and link to it (if they are a product or convolution of the base with something else), or link to the important function if they are a component of a larger/more important function.

### Platforms affected
All.

NickDraper, 2016-06-14:
Assigned to me to split out to authors and volunteers

ElliotAOram, 2016-06-23:
@NickDraper @FedeMPouzols I think Fick Diffusion is already there http://docs.mantidproject.org/nightly/fitfunctions/FickDiffusion.html

FedeMPouzols, 2016-06-23:
@ElliotAOram: true. I might have added it by mistake. I've taken it out.

FedeMPouzols, 2016-07-01:
Reopening this umbrella issue. I have marked in the description the functions that have been added.

PR #16727 (2016-06-29): Documentation for fit function DiffRotDiscreteCircle
Documentation for the elastic and inelastic parts of DiffRotDiscreteCircle has been created. It was necessary to extricate the corresponding fitting functions from DiffRotDiscreteCircle.h(.cpp) into their own header and source files.

Fixes #16687 and partially fixes  #16480

To Test: review the documentation and the code, nothing more is required.

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Rev [... cut]
PR #16730 (2016-07-01): Documentation for fit function DiffSphere
Documentation for the elastic and inelastic parts of DiffSphere has been created. It was necessary to extricate the corresponding fitting functions from DiffSphere.h(.cpp) into their own header and source files.

Fixes #16686 and partially fixes  #16480

To Test: review the documentation and the code, nothing more is required.

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [x] Is the code of a [... cut]
closed 2021-07-27 by stale[bot]