mantidproject/mantid #16509: Remove Qt3Support

FedeMPouzols, 2016-06-01:
This is at the moment blocked by:
 - #16380
 - #16336

which need to be finalized/merged.

Then, clean-up remaining Qt3 issues (http://builds.mantidproject.org/job/master_clean-Qt3-warnings/), and remove the dependency `Qt3Support` from CMakeLists files, and
- in unixes, remove  the dependency`libqt4-qt3support (>= 4.2), on unixes `
- Windows: remove it from the NSIS installer
- Windows: also remove the dll from `ThirdParty`

martyngigg, 2016-06-06:
#16380 has been merged

AndreiSavici, 2016-06-06:
Note for Ubuntu: the following libraries depend on `libqt4-qt3support`: `libqscintilla2-dev`, `libqt4-dev`,`libqt4-dev-bin`, `libqt4-opengl-dev`, `libqtwebkit-dev`, `libqwt5-qt4-dev`, `libqwtplot3d-qt4-dev`

martyngigg, 2016-06-06:
`qwt5` is going to be the next major a headache on the road to Qt 5.

martyngigg, 2016-06-06:
This is quite a good article - https://www.kdab.com/porting-from-qt-4-to-qt-5/

FedeMPouzols, 2016-07-12:
This was closed by  #16560, so the issue can be closed.

closed 2018-06-14 by DavidFair