mantidproject/mantid #16515: Indirect IqtFitMultiple systemtest failure

ElliotAOram, 2016-06-02:
OSIRISIqtandIqtFitMultple system test is currently failing on master see [here](http://builds.mantidproject.org/job/master_systemtests-win7/190/testReport/junit/SystemTests/ISISIndirectInelastic/OSIRISIqtAndIqtFitMulti/) for jenkins output.
This appears to be due to the windows 7 build getting the results previously expected for Linux

NickDraper, 2016-06-02:
Should this be for 3.8?

ElliotAOram, 2016-06-02:
Can be closed as merged with release

ElliotAOram, 2016-06-02:
Reopen so that a more permanent solution can be found.

It is possible that the data and/or fit function are not very good and should be replaced/improved

Spencer has suggested that the issue is that the resolution and sample are constructed from the same run in both the OSIRIS and IRIS case

ElliotAOram, 2016-06-03:
Spencer and Sanghamitra have provided suitable run numbers to replace the ones currently being used.
## IRIS
* Sample = `iris26176_graphite002_red.nxs`
* Resolution = `iris26173_graphite002_res.nxs`

## OSIRIS
* Sample = `osiris97944_graphite002_red.nxs`
* Resolution = `osiris97935_graphite002_res.nxs`

PR #16517 (2016-06-02): Skip OSIRISIqtAndIqtFitMulti test on OSX and Windows
In order to get the builds back on master and to ensure that the builds doesn't fail on the release branch this test is now skipped on both `Windows` and `OSX`

@mantidproject/gatekeepers this is targeted at `release-v3.7` but should probably be merged into master today as well to ensure the nightly builds are restored

**To test:**

* Ensure that the `ISISIndirectInelastic.OSIRISIqtAndIqtFitMulti` is skipped on windows

Fixes #16515

*Does not need to be in the release notes as it is a change to system tests.*

---

#### Reviewer ####

Plea [... cut]
PR #16813 (2016-07-18): Indirect System test - OSIRISIqtAndIqtFitMulti
The OSIRISIqtAndIqtFitMulti system test should now pass on Windows and Linux. It is uncertain if this will work on Mac yet but this can be investigated in #11989 

**To test:**

* Run the ISISIndirectInelastic.OSIRISIqtAndIqtFitMulti system test on windows and Ubuntu and ensure it passes. Ensure it passes on the build servers as well
  * *The new reference files should be downloaded automatically when the system tests are run*

Fixes #16515

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([ful [... cut]
closed 2016-07-18 by FedeMPouzols