mantidproject/mantid #16518: Search Parameter file for corrections in RRO by default

MattKing06, 2016-06-02:
Currently, if no transmission run is given no corrections are applied by default. Scientists would like the algorithm to read the type of corrections from the parameter file (see INTER for an example, where `polynomial` is the defined correction) and use it as a default. If no correction is found in this file, a warning message should be printed.

### Expected behavior

- Reduction will look into the Instrument Parameter file to find what type of correction is needed.

### Actual behavior

- We will only look in the parameter file is 'Auto' is selected.

### Steps to reproduce the behavior

- Run RRO with PolynomialCorrection set to None

### Platforms affected

- All

MattKing06, 2016-06-02:
@raquelalvarezbanos I believe we already doing this in ReflectometryReductionOneAuto see [here](https://github.com/mantidproject/mantid/blob/master/Framework/Algorithms/src/ReflectometryReductionOneAuto.cpp#L406) where 'AutoDetect' is the default for the Correction property. I think the only thing we would need to make this work across all instruments is for there to be a 'correction' tag in each parameter file.

Implementing this AutoDetection in RefelectometryReductionOne doesn't quite make sense for me as the idea behind ReflectometryReductionOne is that no extra 'automatic' properties are supplied to the algorithm.

raquelalvarezbanos, 2016-06-02:
I agree, this issue should be closed (marking it as `Review Required`).

closed 2016-06-07 by NickDraper