mantidproject/mantid #16539: MPI Algorithms broken in v3.7.0

rgmiller, 2016-06-03:
### Expected behavior
MPI framework build succeeds

### Actual behavior
Lots of errors compiling BroadcastWorkspace.cpp and GatherWorkspaces.cpp

At least some of the errors are caused by the fact the declareProperty() wants a unique_ptr now.  Then there are a bunch of "invalid use of incomplete type 'const class Mantid::API::MatrixWorkspace'" errors.  I don't know what that's all about yet.

### Steps to reproduce the behavior
Attempt to build the MPI framework:
module load openmpi-x86_64
scl enable devtoolset-2 "cmake -G 'Unix Makefiles' -DMPI_BUILD=ON -DCMAKE_SKIP_INSTALL_RPATH=ON -DINSTALL_ROOT=/sw/fermi/mantid-mpi $WORKSPACE/Framework"
scl enable devtoolset-2 "cmake --build . --target package -- -j$BUILD_THREADS"

### Platforms affected
Everywhere the MPI build is supposed to work (which I think is just the various Linux distros...)

rgmiller, 2016-06-03:
Note: The MPI build did (and still does) work for v3.6.1.

rgmiller, 2016-06-06:
Note: This is still broken in the 3.7.1 release.

rgmiller, 2016-06-10:
The latest change should fix the compile problems we were having with newer versions of Boost.  This needs some testing, though to make sure I haven't any subtle math errors in GatherWorkspaces.

VickieLynch, 2016-06-13:
I have tested that GatherWorkspaces works for EventWorkspaces and MatrixWorkspaces and the results with multiple mpi tasks matches the serial results.

PR #16641 (2016-06-16): 16539 mpi algorithms dont compile
Fixed up BroadcastWorkspace and GatherWorkspace so that the MPI framework build succeeds.

**To test:**
Vickie Lynch has some test scripts.

Fixes #16539 

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [x] Is the code of an acceptable quality?
- [x] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)? Is it well structured with small focussed classes/methods/functions?
- [x] Are there unit/system tests  [... cut]
closed 2016-06-16 by quantumsteve