mantidproject/mantid #16570: Fix Leak when accessing Help page in ISIS Reflectometry (Polref) interface

MattKing06, 2016-06-08:
### Expected behavior

- No leaks should occur when using the ISIS Reflectometry (Polref) Interface

### Actual behavior

- After using the interface help page and then closing mantid we see this message in the terminal:
```
LEAK: 18 CachedResource
LEAK: 1 Page
LEAK: 1155 RenderObject
LEAK: 1 Frame
LEAK: 1473 WebCoreNode
```

### Steps to reproduce the behavior

- Open the interface (`Interfaces > Reflectometry > ISIS Reflectometry (Polref)`)
- Open the help page for the interface (press the `?` button on the toolbar)
- Close the interface and Mantid
- Notice the aforementioned message in the terminal

### Platforms affected

- All

MattKing06, 2016-06-14:
@raquelalvarezbanos: Am I correct in thinking this affects more than just the ISIS Reflectometry (Polref) interface and is actually part of a wider issue associated with `HelpWindow`?

raquelalvarezbanos, 2016-06-14:
Yes, I think this is a much more general issue: try opening any algorithm dialog and closing Mantid, I've seen the same message in the terminal.

MattKing06, 2016-06-15:
It seems that this is actually a non-issue and is only produced when using a Debug version of Qt, as is used in a Debug Windows build. So this issue is on the Qt side and not ours. See [here](https://bugreports.qt.io/browse/QTBUG-29390) explanation of bug on the Qt side. I am leaving this as "Review Required" in the hope that this issue can be closed.

gemmaguest, 2019-07-24:
I can't reproduce this so I think it is no longer an issue

closed 2019-07-24 by gemmaguest