mantidproject/mantid #16583: out-of-bounds memory access in MDHistoWorkspace::getIsMaskedAt(...)

quantumsteve, 2016-06-09:
### Expected behavior

no out-of-bounds memory access to `MDHistoWorkspace::getIsMaskedAt(...)`

### Actual behavior

Many unit tests pass an invalid index to `MDHistoWorkspace::getIsMaskedAt(...)`

### Steps to reproduce the behavior

run unit tests on the branch associated with this issue.

### Platforms affected

All

quantumsteve, 2016-06-09:
Warning from clang-tidy
```
/Users/builder/jenkins-linode/workspace/clang_tidy/Framework/DataObjects/inc/MantidDataObjects/MDHistoWorkspace.h:237:51: warning: Out of bound memory access (access exceeds upper limit of memory block) [clang-analyzer-alpha.security.ArrayBoundV2]
  bool getIsMaskedAt(size_t index) const { return m_masks[index]; }
                                                  ^
/Users/builder/jenkins-linode/workspace/clang_tidy/Framework/DataObjects/src/MDHistoWorkspace.cpp:388:7: note: Calling 'MDHistoWorkspace::getIsMaskedAt'
  if (this->getIsMaskedAt(linearIndex)) {
      ^
/Users/builder/jenkins-linode/workspace/clang_tidy/Framework/DataObjects/inc/MantidDataObjects/MDHistoWorkspace.h:237:51: note: Out of bound memory access (access exceeds upper limit of memory block)
  bool getIsMaskedAt(size_t index) const { return m_masks[index]; }
```

quantumsteve, 2016-06-09:
Failing unit tests:
```
	350 - PythonInterface_SimpleAPITest (Failed)
	593 - DataObjectsTest_MDHistoWorkspaceIteratorTest (Failed)
	1049 - CurveFittingTest_EvaluateFunctionTest (Failed)
	1175 - CrystalTest_ConnectedComponentLabelingTest (Failed)
	1179 - CrystalTest_FindClusterFacesTest (Failed)
	1191 - CrystalTest_IntegratePeaksHybridTest (Failed)
	1192 - CrystalTest_IntegratePeaksUsingClustersTest (Failed)
	1320 - MDAlgorithmsTest_CompactMDTest (Failed)
	1340 - MDAlgorithmsTest_CutMDTest (Failed)
	1344 - MDAlgorithmsTest_EvaluateMDFunctionTest (Failed)
	1348 - MDAlgorithmsTest_FitMDTest (Failed)
	1358 - MDAlgorithmsTest_IntegrateMDHistoWorkspaceTest (Failed)
	1391 - MDAlgorithmsTest_QueryMDWorkspaceTest (Failed)
	1395 - MDAlgorithmsTest_ReplicateMDTest (Failed)
	1406 - MDAlgorithmsTest_SmoothMDTest (Failed)
	1413 - MDAlgorithmsTest_TransposeMDTest (Failed)
	1416 - MDAlgorithmsTest_WeightedMeanMDTest (Failed)
	1446 - VatesAPITest_MDHWInMemoryLoadingPresenterTest (Failed)
	1447 - VatesAPITest_MDHWLoadingPresenterTest (Failed)
	1451 - VatesAPITest_MetaDataExtractorUtilsTest (Failed)
	1463 - VatesAPITest_MedianAndBelowThresholdRangeTest (Failed)
	1516 - SliceViewerMantidPlotTest_SliceViewerPythonInterfaceTest (Failed)
```

SimonHeybrock, 2016-06-13:
@quantumsteve Have a look at this issue, maybe this is related? https://github.com/mantidproject/mantid/issues/15885.

quantumsteve, 2016-06-13:
@SimonHeybrock Yes, I think that is one of the issues. I'm also observing incorrect values being passed to `MDHistoWorkspaceIterator::jumpTo(size_t)`. In one case, a negative index was cast to an an `size_t`, and in another there appears to be confusion between the number points and number of bins.

DavidFair, 2020-02-20:
Closing as obsolete, if any exist memory sanitizers will pick them up and the issue hasn't been commented out for the last 2 years

closed 2020-02-20 by DavidFair