mantidproject/mantid #16588: Update LoadILLIndirect.cpp for modified entries in new Nexus files for the instrument IN16b

reimundILL, 2016-06-10:
### Expected behavior

Loading IN16b Nexus files

### Resolve discrepancies for new Nexus files

LoadIndirectILL.cpp starting from line 65ff:

     && descriptor.pathExists("/entry0/dataSD/dataSD")         // IN16B
     &&
      descriptor.pathExists(
          "/entry0/instrument/Doppler/doppler_frequency") // IN16B

In Nexus file: /entry0/dataSD/SingleD_data
At the moment no entry doppler_frequency

affects IndirectILLReduction.py when checking .hasProperty('Doppler.doppler_freq')

Perform tests with previous and new IN16b Nexus files

Next issue will be to load IN16b data using Load.cpp

### Platforms affected

All

SpencerHowells, 2016-06-12:
There are 6 Issues waiting to be done for IN16 - 14173, 14179, 14578, 14649, 14650, 14653. These include using the new data format & extra calibration for peak centre.

reimundILL, 2016-06-13:
Thank you for this important information.

PR #16620 (2016-06-23): Re #16588 update of IN16B loader to accommodate changes in the nexus file format
Added the new entry name for single detector data for IN16B indirect reduction.
Check for doppler frequency or mirror sense in the confidence depending on the file format.
Added new test data file with a new IN16B format data.
Added confidence test.
Run load test both on old and new format data.

**To test:**

Load old and new nexus files from IN16B.

Fixes #16588

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/ [... cut]
closed 2016-06-23 by AndreiSavici