mantidproject/mantid #16591: Workspace history speedup

AndreiSavici, 2016-06-10:
This issue is an in-depth look at the performance  problems seen in #16562. PR #16564 is a way to avoid the problem, but we need to find out why history lookup is slow. It seems related to finding the correct loader for files already loaded. The lookup time for a history of a workspace that was processed as part of the group exponentially increases with the number of workspaces in a group. This is unintuitive and has big performance penalty.

NickDraper, 2016-06-13:
This is not to do with a group, it is down to the time creating an algorithm (from history) that has a Filename property that has a long (and lengthening) list of filename in it.

when you call LastAlgorithm(), an algorithm is created and reconsituted with the property values that it held from history.
Most of this is lightning fast, but when you set the Filename property, it:

1. splits the filename into individual files (like this case that had up to 60)
1. for each file finds the correct loader for it (or reports an error if none can be found).

The repeated second step is where things slow down.  However this is currently common code to when you are setting the property normally (not from history) where this validation is important.

NickDraper, 2016-06-13:
If you keep the checks away from using lastAlgorithm (which instantiates the algorithm) and stick to the algorithm history records then this should remain lightning fast.
I'll put in a PR that does this.

PR #16603 (2016-06-30): Speed up inputEnergyMode and add getPropertyValue to AlgHistory
This was a section of code ```inputEnergyMode``` that was slowing down as more complex worksaces were fed to it, and seemed to take too long anyway.

This came out to be that due to the special nature of he Load Algorithm when you reconsitute it by calling history.lastAlgorithm, It verifies and selects a loader for each file in the Filename property, and in this case there were several.

The main change here is to use the algorithm history objects directly rather than calling lastAlgorithm.  This takes the time to run that functio [... cut]
closed 2016-06-30 by FedeMPouzols