mantidproject/mantid #16600: GroupWorkspaces: better default option for input

tom-perkins, 2016-06-13:
*Reported by James Lord*

The parameter *InputWorkspaces* for `GroupWorkspaces` defaults to whatever workspaces were grouped the last time the algorithm was run. These workspaces may not be present any more. It would be more helpful to use the selected workspaces, if any:

> I can select several workspaces in the ADS (workspaces) pane with 
> shift-click or ctrl-click. If I then run GroupWorkspaces, it would be 
> helpful if the "InputWorkspaces" box is filled in with them. But 
> instead it keeps an old default of workspaces I grouped at some point 
> in the past (which aren't there now). As I wanted to group a series of 
> sequentially numbered runs it was easier in the end to write a short 
> script in the Python Window than to type them all in.

tom-perkins, 2016-06-13:
Setting as low priority as this can be easily done with the Group button

NickDraper, 2016-06-14:
This also applies to the MergeRuns algorithm

NickDraper, 2016-06-14:
Started the approach of creating a WorkspaceArryProperty, but this became very troublesome when you start to consider group workspaces, and would have affected much more code.

Starting again with the validator approach (similar to stringlistvalidator)

PR #16655 (2016-06-29): Implement ListProperty widget for workspaces
Simple implemtation of a list property widget for multiple selection of workspaces.

This is indicated by using a alg property std::vector<std::string> with an ADSValidator.

#### Notes: 

1. The ListValidator could be extended later to also support this functiionality

#### Current Limitations :

1. Currently only for Workspaces in the ADS
1. No further validation of the workspaces can be tested other than presence in the ADS

**To test:**

Two algorithms have been modified to use this GUI approach, GroupWorkspaces and MergeRuns

1. try them both o [... cut]
closed 2016-06-29 by FedeMPouzols