mantidproject/mantid #16601: Multi-dataset fitting: global flag lost when adding new function

tom-perkins, 2016-06-13:
*Reported by James Lord*

#### Expected behaviour
If a function parameter is set as global, this flag should not be unset

#### Actual behaviour
Global flag is lost on existing function parameters when a new function is added to the composite function

#### Steps to reproduce the behaviour
- Generate some workspaces
- Open multi-dataset fitting interface and load these workspaces
- Set up a function e.g. `FlatBackground` and set *f0.A0* global
- Add a new function e.g. `LinearBackground`
  - *f0.A0* from the flat background is no longer global

tom-perkins, 2016-06-13:
This is more than just the global flag:

- Ties are also lost when adding a new function, although you can't see this until you change the spectrum or open the "edit parameters" dialog as it stays on screen.

- Values of non-global parameters are reset to those of the first spectrum, rather than keeping their individual values.

This problem is often seen when you run a fit and then want to adjust the fitting function and fit again.

tom-perkins, 2016-06-15:
Other ways to lose the ties and global flags include:
- Loading a new dataset
- Checking or unchecking the NumDeriv box

tom-perkins, 2016-07-27:
I think [this](https://github.com/mantidproject/mantid/blob/6a3e45f40997de8022656fd816172f8c92cb0515/MantidQt/MantidWidgets/src/FunctionBrowser.cpp#L406-L409) is where the global flags/ties are lost - it is called from `attributeChanged`.

tom-perkins, 2016-10-14:
This bug annoys a number of users, aim to fix for 3.9.

Anders-Markvardsen, 2017-03-04:
Hi @AnthonyLim23 moved some tom perkins assigned 3.9 into your space. This one might be for the generic multiple scattering interface it which case it would be worth discussing it with roman. I have put it in temporary holding since I don't recall the muon group bringing this up as urgent in out last meeting with them

closed 2021-02-15 by AnthonyLim23