mantidproject/mantid #16602: MantidNotebook and matplotlib

peterfpeterson, 2016-06-13:
Create a notebook with a cell containing

```python
import matplotlib
%matplotlib notebook
import matplotlib.pyplot as plt
import numpy as np

fig = plt.figure()
x = np.linspace(0, 5, 100)
y = x ** 2
graph = fig.add_axes([0, 0, 1, 0.3]) # left, bottom, width, height (range 0 to 1)

graph.plot(x, y, 'r')

graph.set_xlabel('x')
graph.set_ylabel('y')
graph.set_title('title')
```

### Platforms affected

OSX only

quantumsteve, 2016-06-13:
@peterfpeterson Is this the error message you see?
```
ValueError: Unrecognized backend string "nbagg": valid strings are ['pdf', 'pgf', 'Qt4Agg', 'GTK', 'GTKAgg', 'ps', 'agg', 'cairo', 'MacOSX', 'GTKCairo', 'WXAgg', 'template', 'TkAgg', 'GTK3Cairo', 'GTK3Agg', 'svg', 'WebAgg', 'CocoaAgg', 'emf', 'gdk', 'WX']
```

OS X Yosemite ships with matplotlib version 1.3.1, which doesn't have the `nbagg` backend. The easiest workarounds are installing matplotlib and adjusting your path so it is found before the system one or upgrading to El Capitan (which comes with a newer version of matplotlib).

There was some discussion last year about Mantid requiring a minimum version of matplotlib and either bundling it or creating system packages for that version. I don't think anyone it actively working on it.

peterfpeterson, 2016-06-14:
That is exactly the problem. Thanks.

Sent with Good (www.good.com)

quantumsteve, 2021-02-22:
closing since we package MantidWorkbench with a more recent version of matplotlib.

closed 2021-02-22 by quantumsteve