mantidproject/mantid #16638: Fix coverity issues in DataProcessorUI

raquelalvarezbanos, 2016-06-16:
Report on new defects:

>
2 new defect(s) introduced to mantidproject/mantid found with Coverity Scan.
2 defect(s), reported by Coverity Scan earlier, were marked fixed in the recent build analyzed by Coverity Scan.
>
New defect(s) Reported-by: Coverity Scan
Showing 2 of 2 defect(s)

>
** CID 1356637:  Uninitialized members  (UNINIT_CTOR)
/home/builder/jenkins-linode/workspace/coverity_build_and_submit/MantidQt/MantidWidgets/src/DataProcessorUI/GenericDataProcessorPresenter.cpp: 97 in MantidQt::MantidWidgets::GenericDataProcessorPresenter::GenericDataProcessorPresenter(const MantidQt::MantidWidgets::DataProcessorWhiteList &, const std::map<std::basic_string<char, std::char_traits<char>, std::allocator<char>>, MantidQt::MantidWidgets::DataProcessorPreprocessingAlgorithm, std::less<std::basic_string<char, std::char_traits<char>, std::allocator<char>>>, std::allocator<std::pair<const std::basic_string<char, std::char_traits<char>, std::allocator<char>>, MantidQt::MantidWidgets::DataProcessorPreprocessingAlgorithm>>> &, const MantidQt::MantidWidgets::DataProcessorProcessingAlgorithm &, const MantidQt::MantidWidgets::DataProcessorPostprocessingAlgorithm &)()

>
*** CID 1356637:  Uninitialized members  (UNINIT_CTOR)
/home/builder/jenkins-linode/workspace/coverity_build_and_submit/MantidQt/MantidWidgets/src/DataProcessorUI/GenericDataProcessorPresenter.cpp: 97 in MantidQt::MantidWidgets::GenericDataProcessorPresenter::GenericDataProcessorPresenter(const MantidQt::MantidWidgets::DataProcessorWhiteList &, const std::map<std::basic_string<char, std::char_traits<char>, std::allocator<char>>, MantidQt::MantidWidgets::DataProcessorPreprocessingAlgorithm, std::less<std::basic_string<char, std::char_traits<char>, std::allocator<char>>>, std::allocator<std::pair<const std::basic_string<char, std::char_traits<char>, std::allocator<char>>, MantidQt::MantidWidgets::DataProcessorPreprocessingAlgorithm>>> &, const MantidQt::MantidWidgets::DataProcessorProcessingAlgorithm &, const MantidQt::MantidWidgets::DataProcessorPostprocessingAlgorithm &)()
91                                  "containing commas must be quoted. In case of "
92                                  "conflict between options "
93                                  "specified via this column and options specified "
94                                  "via the <b>Process</b> line edit, the former "
95                                  "prevail.");
96       m_columns = static_cast<int>(m_whitelist.size());
>     CID 1356637:  Uninitialized members  (UNINIT_CTOR)
>     Non-static class member "m_workspaceReceiver" is not initialized in this constructor nor in any functions that it calls.
97     }
98
99     /**
100     * Destructor
101     */
102     GenericDataProcessorPresenter::~GenericDataProcessorPresenter() {}
>
** CID 1356636:  API usage errors  (INVALIDATE_ITERATOR)
/home/builder/jenkins-linode/workspace/coverity_build_and_submit/MantidQt/MantidWidgets/src/DataProcessorUI/ParseKeyValueString.cpp: 38 in MantidQt::MantidWidgets::parseKeyValueString(const std::basic_string<char, std::char_traits<char>, std::allocator<char>>&)()

>
*** CID 1356636:  API usage errors  (INVALIDATE_ITERATOR)
/home/builder/jenkins-linode/workspace/coverity_build_and_submit/MantidQt/MantidWidgets/src/DataProcessorUI/ParseKeyValueString.cpp: 38 in MantidQt::MantidWidgets::parseKeyValueString(const std::basic_string<char, std::char_traits<char>, std::allocator<char>>&)()
32
33         if (valVec.size() > 1) {
34           // We split on all '='s. The first delimits the key, the rest are assumed
35           // to be part of the value
36           std::string key = valVec[0];
37           // Drop the key from the values vector
>     CID 1356636:  API usage errors  (INVALIDATE_ITERATOR)
>     Using invalid iterator "valVec.begin()".
38           valVec.erase(valVec.begin());
39           // Join the remaining sections,
40           std::string value = boost::algorithm::join(valVec, "=");
41
42           // Remove any unwanted whitespace
43           boost::trim(key);

MattKing06, 2016-06-16:
@raquelalvarezbanos I've already opened up an issue associated with the ParseKeyValueString here #16625

raquelalvarezbanos, 2016-06-16:
Ah, OK, thanks Matt, I didn't see it, therefore this one can be closed.

PR #16643 (2016-06-21): Fix Coverity issue in GenericDataProcessorPresenter
In #16638 it is evident that the `m_workspaceReceiver` member of GenericDataProcessorPresenter.cpp is not initialized upon construction. This needs to be changed so that `m_workspaceReceiver` is initialized when a GenericDataProcessorPresenter is created.

**To test:**

- Code Review

Fixes #16638

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [x] Is the code of an acceptable quality?
- [x] Does the code conform to the [coding  [... cut]
closed 2016-06-21 by FedeMPouzols