mantidproject/mantid #16649: LoadNexus algorithm fails when loading a saved nexus file after AppendSpectra

shahrozahmed1, 2016-06-17:
When you run the following Fit documentation test:

```
ws = CreateSampleWorkspace(Function="User Defined", UserDefinedFunction="name=LinearBackground, \
   A0=0.3;name=Gaussian, PeakCentre=5, Height=10, Sigma=0.7", NumBanks=1, BankPixelWidth=1, XMin=0, XMax=10, BinWidth=0.1)

workspaceIndex = 0
startX = 1
endX = 9

tryCentre = '4'
sigma = '1'
height = '8'
myFunc = 'name=Gaussian, Height='+height+', PeakCentre='+tryCentre+', Sigma='+sigma

fitStatus, chiSq, covarianceTable, paramTable, fitWorkspace = Fit(InputWorkspace='ws', \
   WorkspaceIndex=0, StartX = startX, EndX=endX, Output='fit', Function=myFunc)
   
```

Saving and Loading the `Fit_Workspace` as nexus file work fine.
However if you use `appendSpectra` algorithm e.g.:
`AppendSpectra(InputWorkspace1 = "fit_Workspace", InputWorkspace2 = "fit_Workspace", OutputWorkspace = "test_append")`
And then if try to Save and Load `test_append` workspace, 
```
SaveNexus(InputWorkspace = "test_append", Filename="C:/Mantid/test_append")
LoadNexus(Filename="C:/Mantid/test_append", OutputWorkspace = "test_load")
```
The Load algorithm fails and displays the following error message:
```
LoadNexus started
Error in execution of algorithm LoadNexusProcessed:
Attempt to load from an empty dataset /mantid_workspace_1/workspace/axis2
Error in execution of algorithm LoadNexus:
Attempt to load from an empty dataset /mantid_workspace_1/workspace/axis2
```

DavidFair, 2016-09-16:
Related to #17479

DavidFair, 2016-09-29:
Closing as duplicate of #17479 - the reason this is failing Mantid seems to think a y axis in certain conditions holds only text data so doesn't save out correctly to the nxs file. This is why we are seeing the loader subsequently fail.

closed 2016-09-29 by DavidFair