mantidproject/mantid #16651: CalculateFlatBackground - various correctness issues

SimonHeybrock, 2016-06-17:
`CalculateFlatBackground` has several issues, severity differs:

1. When used in "Mean" mode the variance computation divides by `(numBins * numBins)`. It should be `numBins`. As a consequence, the error is vastly underestimated.
2. "Mean" does not weigh values by their error, i.e., a single outlier with a huge error bar can dominate the mean value and make the background value meaningless.
3. "Mean" does not take into account the bin width. An area with many tiny bins will dominate the result.
4. In "Linear Fit" mode the error is not taken into account, despite being available from the fit result.
5. However, in "Linear Fit" mode errors may get modified for "NullifyNegative" and bins with negative values. That is, error bars for all but the negative bins are unchanged.
6. The input indices are not verified. If an index shows up twice in the list, the background subtraction will be run twice for that spectrum.

All these were found in the code, I have not attempted to demonstrate the issues by using the algorithm.

To solve this:
- Why do we actually need Linear Fit and then take the center value? Why not just a constant fit?
- Why not use Fit with a constant instead of the manually and wrongly calculated mean?

NickDraper, 2016-06-22:
@abuts Alex, Is this used in the Direct Inelastic workflow?
@mantid-roman Do you have any views  on this?

SimonHeybrock, 2016-06-22:
I will leave this unassigned until we have figured out how to fix this. It does not make sense if I fix this in a way that is incompatible with requirements I am not aware of.

mantid-roman, 2016-06-22:
I am not familiar well with the algorithm but it seems to me that the mean and its variance are calculated correctly. 

The outliers can have much smaller values than average and with Poisson distribution weighing them by their errors will increase their contributions. I don't think it's a solution.

As I understand it the real issue here is if the algorithm doesn't do a good job. Concrete examples are needed which would test the solution.

I have always said that fitting to histograms is a bad idea.

SimonHeybrock, 2016-06-22:
@mantid-roman No, the standard deviation should decrease with the sqrt of the number of measurements. See, e.g., https://en.wikipedia.org/wiki/Standard_deviation.

mantid-roman, 2016-06-22:
It's not the standard deviation, it's the variance of the mean: https://onlinecourses.science.psu.edu/stat414/node/167

SimonHeybrock, 2016-06-22:
@mantid-roman Ok, after checking the formula, the factor `numbins * numbins` is correct. It is equivalent to what I wrote above, if you start with a measurement of standard deviation `sigma`, the standard deviation of the mean decreases with the sqrt of the number of measurements. Basically what we are doing here is approximating the true `sigma` by the mean of standard deviations of all measurements, this accounts for the first factor `numbins`. The second factor `numbins` is then the one that gives the `1/sqrt(N)` dependence for the standard deviation of the mean.

closed 2021-03-03 by stale[bot]