mantidproject/mantid #16665: Spectrum Viewer plots wrong Y axis (if axis is replaced)

tom-perkins, 2016-06-20:
#### Expected behaviour
If the Y axis of a workspace is replaced, plotting the workspace should use the new axis

#### Actual behaviour
Spectrum Viewer still uses the old axis values

#### Steps to reproduce the behaviour
```python
ws = CreateSampleWorkspace()
na = NumericAxis.create(200)
for i in range(200):
    na.setValue(i, (i-200.0))
ws.replaceAxis(1, na)
```
- Compare the results of right-clicking and choosing "Color Fill Plot" vs. "Show Spectrum Viewer"
- Color fill (correct):
[image]

- Spectrum viewer (wrong):
[image]

NickDraper, 2016-06-22:
Very niche issue in a niche tool.

Put as low priority for R3.9, by then we should have some usage data, we may find that this tool is never used.

NickDraper, 2016-10-14:
archived for now

closed 2016-10-14 by NickDraper