mantidproject/mantid #16666: Curve labels wrong for bin edge Y data

tom-perkins, 2016-06-20:
#### Expected behaviour
When plotting spectra from a workspace where the Y axis is bin edges, curves should be labelled with bin centre values

#### Actual behaviour
If Y axis is bin edges, the spectra are labelled with the lower edge of their bin and not the centre value

#### Steps to reproduce the behaviour
- Create a workspace with a `BinEdgeAxis` as the Y axis:
```python
tw=WorkspaceFactory.create("Workspace2D",XLength=5,YLength=5,NVectors=5)
for i in range(5):
	for j in range(5):
		tw.dataX(i)[j]=(j-2.0)*1.0
		tw.dataY(i)[j]=(i-2.0)**2+(j-2.0)**2
na = BinEdgeAxis.create(6)
for i in range(6):
	na.setValue(i,(i-2.5)-0.05*(i-2.5)**3)
tw.replaceAxis(1,na)

AnalysisDataService.addOrReplace("tw",tw)
```
- Right-click and plot spectrum > plot all
- The legend labels the curves with their lower bin edges, not the centre values:

[image]

NickDraper, 2016-06-22:
BinEdge Axes are pretty rare

tom-perkins, 2016-10-12:
Duplicate of #17782

closed 2017-01-16 by NickDraper