mantidproject/mantid #16679: Fix WeightedLeastSquaresTest for GSL2

FedeMPouzols, 2016-06-21:
Tests failing with GSL2: http://builds.mantidproject.org/view/All/job/master_clean-archlinux/405/#showFailuresLink
Fix:
- SystemTests.WeightedLeastSquaresTest.WeightedLeastSquaresTest

This system test may also need revisiting now that we have more fitting tetss in place.

NickDraper, 2016-09-13:
Federico, It would be good to get this into the release, I've marked high priority for now, please give me an update when you can

FedeMPouzols, 2016-09-16:
@NickDraper: as I'm working on related fitting tests stuff I'll keep this mind and see if I find a good fix for it. There are several other tests that fail, and some of them are proving really hard to fix, see for example #17245.

rosswhitfield, 2016-09-30:
Fixed in #17642

PR #17642 (2016-09-30): Fix WeightedLeastSquaresTest for GSL2
Removed a test case which shouldn't be there in the first place. Increased tolerance for another test.

**To test:**

System tests must pass.

Fixes #16679.

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Does the code conform to the [coding standards](http://www.mantidproject.org/Coding_Standards)? Is it well structured with small focussed classes/me [... cut]
closed 2016-09-30 by rosswhitfield