mantidproject/mantid #16680: Fix system unit/doc/system tests that fail with GSL2

FedeMPouzols, 2016-06-21:
After maintenance work for release 3.8 Mantid compiles and works with GSL2, but 9 system tests, 1 doc test, and 1 unit test still fail:
http://builds.mantidproject.org/view/All/job/master_clean-archlinux/405/#showFailuresLink

This needs fixing for Mantid to work on Ubuntu 16. Fix them:
- Engineeringg: #16678
- Fitting tests: #16679
- ISIS Indirect Inelastic: TODO
- SNS/other: TODO

rosswhitfield, 2016-06-21:
There is also one sporadically failing unit test `CurveFittingTest.MultiDomainFunctionTest.test_Fit_algorithm`. See [here](http://builds.mantidproject.org/job/master_clean-ubuntu-16.04/119/)

peterfpeterson, 2016-06-21:
@rosswhitfield Please create a ticket and add it to the list above.

quantumsteve, 2016-09-02:
GSL 2.2 breaks yet another unit test. The results below are from OS X.

```
$ ctest -R CurveFittingTest_DiffRotDiscreteCircleTest  --verbose
UpdateCTestConfiguration  from :/Users/svh/Documents/MantidProject/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/Users/svh/Documents/MantidProject/build/DartConfiguration.tcl
Test project /Users/svh/Documents/MantidProject/build
Constructing a list of tests
Done constructing a list of tests
Checking test dependency graph...
Checking test dependency graph end
test 1154
    Start 1154: CurveFittingTest_DiffRotDiscreteCircleTest

1154: Test command: /usr/local/Cellar/cmake/3.6.1/bin/cmake "-E" "chdir" "/Users/svh/Documents/MantidProject/build/bin/Testing" "/Users/svh/Documents/MantidProject/build/bin/CurveFittingTest" "DiffRotDiscreteCircleTest"
1154: Test timeout computed to be: 300
1154: FrameworkManager-[Notice] Welcome to Mantid 3.7.20160902.2041
1154: FrameworkManager-[Notice] Please cite: http://dx.doi.org/10.1016/j.nima.2014.07.029 and this release: http://dx.doi.org/10.5286/Software/Mantid
1154: Running 9 tests.Fit-[Notice] Fit started
1154: Fit-[Notice] Fit successful, Duration 0.01 seconds
1154: .Fit-[Notice] Fit started
1154: Fit-[Notice] Fit successful, Duration 0.01 seconds
1154: .Fit-[Notice] Fit started
1154: Fit-[Notice] Fit successful, Duration 0.02 seconds
1154: .Fit-[Notice] Fit started
1154: Fit-[Notice] Fit successful, Duration 0.02 seconds
1154: .Fit-[Notice] Fit started
1154: Fit-[Notice] Fit successful, Duration 0.00 seconds
1154: ...Fit-[Notice] Fit started
1154: Fit-[Notice] Fit successful, Duration 0.63 seconds
1154: 
1154: In DiffRotDiscreteCircleTest::testDiffRotDiscreteCircle:
1154: /Users/svh/Documents/MantidProject/mantid/Framework/CurveFitting/test/Functions/DiffRotDiscreteCircleTest.h:298: Error: Expected (fitalg_structure_factor->getParameter("Radius") == 1.567) up to 1.567 * 0.05 (0.0783), found (1.7477 != 1.5670)
1154: /Users/svh/Documents/MantidProject/mantid/Framework/CurveFitting/test/Functions/DiffRotDiscreteCircleTest.h:300: Error: Expected (fitalg_structure_factor->getParameter("Decay") == 7.567) up to 7.567 * 0.05 (0.3783), found (9.1182 != 7.5670)
1154: Failed 1 of 9 tests
1154: Success rate: 88%
1/1 Test #1154: CurveFittingTest_DiffRotDiscreteCircleTest ...***Failed    1.07 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   1.17 sec

The following tests FAILED:
	1154 - CurveFittingTest_DiffRotDiscreteCircleTest (Failed)
Errors while running CTest
```

rosswhitfield, 2016-09-06:
The test doesn't fail on Arch Linux using GSL `2.2.1`

rosswhitfield, 2017-05-17:
@quantumsteve Have you tried GSL 2.3?

DavidFair, 2018-06-14:
Closing as all test issues have been resolved

closed 2018-06-14 by DavidFair