mantidproject/mantid #16681: Fix ISIS Indirect Inelastic system tests for GSL2

FedeMPouzols, 2016-06-21:
Tests failing with GSL2: http://builds.mantidproject.org/view/All/job/master_clean-archlinux/405/#showFailuresLink

Fix the Indirect Inelastic related ones (or split into more issues/tickets):
- SystemTests.ISISIndirectInelastic.IRISIqtAndIqtFit
- SystemTests.ISISIndirectInelastic.OSIRISIqtAndIqtFit
- SystemTests.ISISIndirectInelastic.IRISConvFit
- SystemTests.ISISIndirectInelastic.IRISIqtAndIqtFitMulti
- SystemTests.ISISIndirectInelastic.OSIRISConvFit
- SystemTests.ISISIndirectInelastic.OSIRISIqtAndIqtFitMulti

FedeMPouzols, 2016-06-21:
@ElliotAOram, @martyngigg: I'd guess you're the ones that have a better handle on these tests?

SimonHeybrock, 2016-06-22:
@FedeMPouzols @ElliotAOram @martyngigg This is a long shot, but check out this issue regarding `CalculateFlatBackground`: #16651. The algorithm seems to be used in the ISIS indirect code, and the issue implies that the error bar are too small, so that might influence fitting.

NickDraper, 2016-09-13:
Ask Spencer to help on this

peterfpeterson, 2016-09-28:
The PR fixed this. GitHub didn't close it for some reason.

rosswhitfield, 2016-09-28:
Fixed in #17608

PR #17608 (2016-09-27): Fix isis indirect inelastic system tests gsl2
Improved stability of the fits and increased the tolerance of results comparison. Found that one of the reason for differences in results of calculations is the VS optimisation. 

**To test:**

Check that the ISIS indirect inelastic system tests pass on a system using the GSL2.

Fixes #16681.

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable quality?
- [ ] Doe [... cut]
closed 2016-09-28 by peterfpeterson