mantidproject/mantid #16687: Documentation for fit function DiffRotDiscreteCircle

jmborr, 2016-06-21:
Includes documentation for the elastic (ElasticDiffRotDiscreteCircle) and inelastic (InelasticDiffRotDiscreteCircle) parts

Adresses in part issue #16480

jmborr, 2016-06-23:
@FedeMPouzols There is a problem when building the documentation:  `ElasticDiffRotDiscreteCircle`, `InelasticDiffRotDiscreteCircle`, and `DiffRotDiscreteCircle` are all implemented in `DiffRotDiscreteCircle.cpp` because the latter is a composite of the first two.

Building the documentation with:
`bin/MantidPlot -xq docs/runsphinx_html.py -R ElasticDiffRotDiscreteCircle`
will search for source ElasticDiffRotDiscreteCircle.cpp, but only DiffRotDiscreteCircle.cpp exists so no documentation is built.

FedeMPouzols, 2016-06-24:
I see. I think that the recommended approach would be one separate file per function? @martyngigg? @NickDraper?

martyngigg, 2016-06-24:
If they are visible enough to be distinct functions then yes they should be in separate files

jmborr, 2016-06-24:
OK, I'll do that.

PR #16727 (2016-06-29): Documentation for fit function DiffRotDiscreteCircle
Documentation for the elastic and inelastic parts of DiffRotDiscreteCircle has been created. It was necessary to extricate the corresponding fitting functions from DiffRotDiscreteCircle.h(.cpp) into their own header and source files.

Fixes #16687 and partially fixes  #16480

To Test: review the documentation and the code, nothing more is required.

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Rev [... cut]
closed 2016-06-29 by SimonHeybrock