mantidproject/mantid #16737: Include PolarizationCorrection in ReflectometryReductionOne

MattKing06, 2016-06-27:
Issue reported by Users: 
```
The workflow diagram does not document the calculations performed before outputting the workspace
in wavelength.
Specifically they pointed out that background subtraction and polarization corrections are missing
```

The `PolarizationCorrection` step is only present in the RROA algorithm. Since RROA is supposed to be a wrapper around RRO that will grab properties from the IDF when possible, it seems only right in terms of continuity that `PolarizationCorrection` should be a part of RRO; and users should be able to manually set the properties needed for this correction algorithm

MattKing06, 2016-06-27:
@OwenArnold @raquelalvarezbanos 

After a very short investigation it seems that we will not be able to implement Polarization corrections in RRO without redesigning how RRO deals with WorkspaceGroups. I'd forgotten that the input workspaces for `PolarizationCorrection` were grouped and thus are only used when the input workspace for RROA is a WorkspaceGroups. 

We will have to create a method similar to `ProcessGroups()` in RRO that will allow us to reduce group workspace and thus perform polarization corrections.

However, in an effort to harmonize the two algorithms I will update the documentation of RRO to mention that RROA will auto-populate some properties and also perform extra corrections so that hopefully the confusion is minimal until this change can be brought in.

raquelalvarezbanos, 2016-11-24:
Having a better understanding of the algorithms, I think it does not make sense to move `PolarizationAnalysis` to `ReflectometryReductionOne`, therefore I am closing this issue.

closed 2016-11-24 by raquelalvarezbanos