mantidproject/mantid #16739: Update Python algorithms & functions for compatibility with both Python 2 & 3

martyngigg, 2016-06-27:
A cheat sheet is available at http://python-future.org/compatible_idioms.html that details the differences between Python 2 & 3 code. Most of our incompatibilities fall in to one of three categories:

* `print` is now a function and requires parentheses in Python 3
* commas not allowed in `except` clauses.
* `dict.iteritems()` no longer exists 

We need to maintain compatibility with Python 2. The `six` module (https://pythonhosted.org/six/) provides functionality to support both versions and the above cheat sheet demonstrates how to use it in various situations. For dictionary iteration use

```python
from six import iteritems
...

for k, v in iteritems(test_dict):
```

This is an umbrella issue. Please add your username to the list below and include this ticket number in your PR.

1. 
2. 
3. 
4. 
5. 
6. 
7. 
8. 
9.  

Tasks for each file:

* Check if there is a test for the algorithm/function. If not contact the original author if possible and work with them to implement one.
* Add the line `from __future__ import (absolute_import, division, print_function)` as the **first** import of the file
* Run the test
* Fix issues that arise using the cheat sheet.

**Group 1**
PythonInterface/plugins/algorithms/AlignComponents.py
PythonInterface/plugins/algorithms/ApplyNegMuCorrection.py
PythonInterface/plugins/algorithms/BASISReduction311.py
PythonInterface/plugins/algorithms/BASISReduction.py
PythonInterface/plugins/algorithms/CalculateSampleTransmission.py
PythonInterface/plugins/algorithms/CalibrateRectangularDetectors.py
PythonInterface/plugins/algorithms/CheckForSampleLogs.py
PythonInterface/plugins/algorithms/CleanFileCache.py
PythonInterface/plugins/algorithms/CollectHB3AExperimentInfo.py
PythonInterface/plugins/algorithms/CompareSampleLogs.py
PythonInterface/plugins/algorithms/ComputeCalibrationCoefVan.py
PythonInterface/plugins/algorithms/ComputeIncoherentDOS.py
PythonInterface/plugins/algorithms/ConjoinFiles.py
PythonInterface/plugins/algorithms/ConjoinSpectra.py
PythonInterface/plugins/algorithms/ConvertSnsRoiFileToMask.py
PythonInterface/plugins/algorithms/CorrectLogTimes.py
PythonInterface/plugins/algorithms/CorrectTOF.py
PythonInterface/plugins/algorithms/CreateCacheFilename.py
PythonInterface/plugins/algorithms/CreateEmptyTableWorkspace.py
PythonInterface/plugins/algorithms/CreateLeBailFitInput.py
PythonInterface/plugins/algorithms/CylinderPaalmanPingsCorrection2.py

**Group 2**
PythonInterface/plugins/algorithms/DakotaChiSquared.py
PythonInterface/plugins/algorithms/DNSComputeDetEffCorrCoefs.py
PythonInterface/plugins/algorithms/dnsdata.py
PythonInterface/plugins/algorithms/DNSFlippingRatioCorr.py
PythonInterface/plugins/algorithms/DNSMergeRuns.py
PythonInterface/plugins/algorithms/DPDFreduction.py
PythonInterface/plugins/algorithms/DSFinterp.py
PythonInterface/plugins/algorithms/EnggCalibrateFull.py
PythonInterface/plugins/algorithms/EnggCalibrate.py
PythonInterface/plugins/algorithms/EnggFitDIFCFromPeaks.py
PythonInterface/plugins/algorithms/EnggFitPeaks.py
PythonInterface/plugins/algorithms/EnggFocus.py
[... 140 lines cut ...]
PythonInterface/plugins/algorithms/WorkflowAlgorithms/SANSStitch.py
PythonInterface/plugins/algorithms/WorkflowAlgorithms/SavePlot1D.py
PythonInterface/plugins/algorithms/WorkflowAlgorithms/SimulatedDensityOfStates.py
PythonInterface/plugins/algorithms/WorkflowAlgorithms/SofQWMoments.py
PythonInterface/plugins/algorithms/WorkflowAlgorithms/SwapWidths.py
PythonInterface/plugins/algorithms/WorkflowAlgorithms/TimeSlice.py
PythonInterface/plugins/algorithms/WorkflowAlgorithms/TOSCABankCorrection.py

**Group 9**
PythonInterface/plugins/algorithms/WorkflowAlgorithms/TransformToIqt.py
PythonInterface/plugins/algorithms/WorkflowAlgorithms/TransmissionUtils.py
PythonInterface/plugins/algorithms/WorkflowAlgorithms/USANSReduction.py
PythonInterface/plugins/algorithms/WorkflowAlgorithms/VesuvioDiffractionReduction.py
PythonInterface/plugins/functions/ChudleyElliot.py
PythonInterface/plugins/functions/DSFinterp1DFit.py
PythonInterface/plugins/functions/Examples/Example1DFunction.py
PythonInterface/plugins/functions/Examples/ExamplePeakFunction.py
PythonInterface/plugins/functions/FickDiffusion.py
PythonInterface/plugins/functions/GuinierPorod.py
PythonInterface/plugins/functions/Guinier.py
PythonInterface/plugins/functions/HallRoss.py
PythonInterface/plugins/functions/Lorentz.py
PythonInterface/plugins/functions/Porod.py
PythonInterface/plugins/functions/StretchedExpFT.py
PythonInterface/plugins/functions/TeixeiraWater.py

martyngigg, 2016-07-20:
Group 1 was fixed in #16814

rosswhitfield, 2016-07-20:
I've moved `PythonInterface/plugins/algorithms/dnsdata.py` from group 2 to 3 because LoadDNSLegacy is the only thing that uses it so I will fix it at the same time.

rosswhitfield, 2016-07-20:
Is the plan to fix the unittests for python3 after all the algorithms have been made compatible? Or should I be trying to fix them in the same pull request?

martyngigg, 2016-07-21:
I'd say try and fix them in the same pull request.

mganeva, 2016-07-21:
Group 6: PythonInterface/plugins/algorithms/TOFTOFConvertTofToDeltaE.py is deprecated and can be removed in the next release.

rosswhitfield, 2016-08-17:
@martyngigg We have some regression with the python 3 unit tests, I see a lot of them are now segfaulting when the previously didn't. It looks like the ones that use `validateInputs`. Any idea?

```
385 - PythonInterface_SimpleAPITest (SEGFAULT)
489 - PythonAlgorithms_AddSampleLogMultipleTest (SEGFAULT)
490 - PythonAlgorithms_AlignComponentsTest (SEGFAULT)
491 - PythonAlgorithms_ApplyPaalmanPingsCorrectionTest (SEGFAULT)
494 - PythonAlgorithms_CalculateSampleTransmissionTest (SEGFAULT)
497 - PythonAlgorithms_CompareSampleLogsTest (SEGFAULT)
498 - PythonAlgorithms_ComputeCalibrationCoefVanTest (SEGFAULT)
501 - PythonAlgorithms_DetectorFloodWeightingTest (SEGFAULT)
502 - PythonAlgorithms_IndirectCalibrationTest (SEGFAULT)
503 - PythonAlgorithms_CorrectTOFTest (SEGFAULT)
505 - PythonAlgorithms_CylinderPaalmanPingsCorrection2Test (SEGFAULT)
507 - PythonAlgorithms_DNSComputeDetEffCorrCoefsTest (SEGFAULT)
508 - PythonAlgorithms_DNSMergeRunsTest (SEGFAULT)
509 - PythonAlgorithms_DNSFlippingRatioCorrTest (SEGFAULT)
512 - PythonAlgorithms_EnggCalibrateTest (SEGFAULT)
513 - PythonAlgorithms_EnggFitDIFCFromPeaksTest (SEGFAULT)
515 - PythonAlgorithms_EnggFocusTest (SEGFAULT)
517 - PythonAlgorithms_ExportSpectraMaskTest (Failed)
522 - PythonAlgorithms_FlatPlatePaalmanPingsCorrectionTest (SEGFAULT)
526 - PythonAlgorithms_IndirectAnnulusAbsorptionTest (SEGFAULT)
527 - PythonAlgorithms_IndirectCylinderAbsorptionTest (SEGFAULT)
528 - PythonAlgorithms_IndirectFlatPlateAbsorptionTest (SEGFAULT)
529 - PythonAlgorithms_IndirectILLReductionTest (SEGFAULT)
530 - PythonAlgorithms_IndirectResolutionTest (SEGFAULT)
533 - PythonAlgorithms_ILLIN16BCalibrationTest (SEGFAULT)
534 - PythonAlgorithms_IqtFitMultipleTest (SEGFAULT)
535 - PythonAlgorithms_IqtFitSequentialTest (SEGFAULT)
536 - PythonAlgorithms_ISISIndirectDiffractionReductionTest (SEGFAULT)
537 - PythonAlgorithms_ISISIndirectEnergyTransferTest (SEGFAULT)
544 - PythonAlgorithms_LoadNMoldyn3AsciiTest (SEGFAULT)
545 - PythonAlgorithms_LoadNMoldyn4Ascii1DTest (SEGFAULT)
546 - PythonAlgorithms_LoadNMoldyn4AsciiTest (SEGFAULT)
547 - PythonAlgorithms_MaskAngleTest (SEGFAULT)
552 - PythonAlgorithms_MolDynTest (Failed)
553 - PythonAlgorithms_MSDFitTest (SEGFAULT)
556 - PythonAlgorithms_OSIRISDiffractionReductionTest (SEGFAULT)
557 - PythonAlgorithms_ResNorm2Test (SEGFAULT)
559 - PythonAlgorithms_SANSFitShiftScaleTest (Failed)
561 - PythonAlgorithms_SaveNexusPDTest (SEGFAULT)
562 - PythonAlgorithms_SavePlot1DTest (SEGFAULT)
565 - PythonAlgorithms_SimulatedDensityOfStatesTest (SEGFAULT)
571 - PythonAlgorithms_Stitch1DManyTest (Failed)
574 - PythonAlgorithms_SuggestTibHYSPECTest (Failed)
576 - PythonAlgorithms_SymmetriseTest (SEGFAULT)
577 - PythonAlgorithms_UpdatePeakParameterTableValueTest (Failed)
578 - PythonAlgorithms_SANSDarkRunBackgroundCorrectionTest (SEGFAULT)
580 - PythonAlgorithms_TimeSliceTest (SEGFAULT)
581 - PythonAlgorithms_TOFTOFConvertTofToDeltaETest (SEGFAULT)
582 - PythonAlgorithms_TOFTOFCropWorkspaceTest (SEGFAULT)
583 - PythonAlgorithms_TOFTOFMergeRunsTest (Failed)
584 - PythonAlgorithms_TOSCABankCorrectionTest (SEGFAULT)
585 - PythonAlgorithms_TransformToIqtTest (SEGFAULT)
586 - PythonAlgorithms_ExportSampleLogsToCSVFileTest (Failed)
587 - PythonAlgorithms_ExportExperimentLogTest (Failed)
589 - PythonAlgorithms_VesuvioDiffractionReductionTest (SEGFAULT)
590 - PythonAlgorithms_VesuvioPreFitTest (Failed)
591 - PythonAlgorithms_VesuvioResolutionTest (SEGFAULT)
592 - PythonAlgorithms_VesuvioThicknessTest (Failed)
593 - PythonAlgorithms_VesuvioTOFFitTest (Failed)
```

martyngigg, 2016-08-17:
@rosswhitfield I'm not sure off the top of my head. There was an issue with `3` where you could no longer use a `dict` without holding the global interpreter lock and `validateInputs()` does return a `dict` instance. I wonder if this is to do with my changes around the exception propagation?

rosswhitfield, 2016-10-11:
All the groups are now complete **but** we have new algorithms added since this issues was created. So ...

**Group 10**
PythonInterface/plugins/algorithms/AngularAutoCorrelationsSingleAxis.py
PythonInterface/plugins/algorithms/AngularAutoCorrelationsTwoAxes.py
PythonInterface/plugins/algorithms/ExportSpectraMask.py
PythonInterface/plugins/algorithms/LoadPreNexusLive.py
PythonInterface/plugins/algorithms/VelocityAutoCorrelations.py
PythonInterface/plugins/algorithms/VelocityCrossCorrelations.py
PythonInterface/plugins/algorithms/WorkflowAlgorithms/BayesStretch.py
PythonInterface/plugins/algorithms/WorkflowAlgorithms/SANSPatchSensitivity.py

PR #17872 (2016-10-24): Update algorithms for Python3 - Group 10
This is Group 10 of #16739 and with this I think we should close #16739 

**To test:**
* Code review
* Compile with python3 and check that the related unit tests pass 

After this we still have some sporadically failing unit tests with python 3. With this PR you should find all `PythonAlgorithms` test passing with the occasional exception of the following.
```
	514 - PythonAlgorithms_EnggCalibrateFullTest (Failed)
	576 - PythonAlgorithms_StatisticsOfTableWorkspaceTest (Failed)
	598 - PythonAlgorithms_VesuvioResolutionTest (Failed)
	600 - PythonAlgorithm [... cut]
closed 2016-10-24 by martyngigg