mantidproject/mantid #16770: Add custom conversion option to UnitConversion

NickDraper, 2016-07-01:
### Expected behavior
Unit  conversions that cannot be done through ConvertUnits but can be defined using an equation can be applied to the data.

### Actual behavior
Currently we have ConvertUints that only supports predefined units and links between them, and ConvertAxisByFormula that cannot handle any geometry aspect in the transformation.

### Steps to reproduce the behavior

Ansto have a dataset for which there is no sensible L1, but want to convert from wavelength to Momentum Transfer.

This is likely to come up more at reactor sources.
@ianbush Heads up!

### Platforms affected
All

NickDraper, 2016-07-01:
@AntonPiccardoSelg This is the ticket we discussed

SpencerHowells, 2016-07-01:
I've been struggling with this for years dealing with ILL diffraction data! Converting from angle to Q & vice versa.

NickDraper, 2016-07-08:
This should be ready for testing soon, I've just submitted the pull request.

PR #16824 (2016-08-03): ConvertAxisByFormula add geometry handling
Improved the abilities of ConvertAxisByFormula by adding geometry support.  This will be particularly helpful for reactor sources where our unit conversions will not support what they need to do.

**To test:**

1. unit tests work
1. read the changed algorithm docs
1. play with various forumlae.
1. try this out.

``` python 
   wsWavelength = CreateSampleWorkspace(XUnit='Wavelength', XMin=2, XMax=6, BinWidth=0.05)
   # Convert to momentum transfer
   # directly using a formula
   wsMTbyFormula = ConvertAxisByFormula(InputWorkspace=wsWavelength,  Formul [... cut]
closed 2016-08-03 by peterfpeterson