mantidproject/mantid #16792: Spectrum Viewer display in Radians

LamarMoore, 2016-07-05:
*This issue is based on feedback I received on the Mantid training course held at the MLZ in Munich June 30 - July 1.*

The `Image Info` tab of the Spectrum Viewer contains angles in radians which has proven useless to some of the instrument scientists. They would prefer if angles were displayed in degrees as was the case in previous versions of Mantid.

Dr Marina Ganeva of MLZ stated:

*"Radians as units make this information completely useless, because the displayed angle in radians will be the same for 50 detectors."*

[image]

AndreiSavici, 2016-07-05:
It is not in radians. I tried to run it on CNCS_7860, with largest two theta around 130 degrees, but on spectrum viewer I could not get a two theta greater than 0.05

mganeva, 2016-07-05:
I've checked, for TOFTOF is the same: it is not radians. For radians the situation would not be that bad. However, degrees would provide much better accuracy.

AndreiSavici, 2016-07-05:
https://github.com/mantidproject/mantid/blob/c3e2386103a6b7080489e65cc50e6383e149d2be/MantidQt/SpectrumViewer/src/MatrixWSDataSource.cpp#L289 needs to be rad2deg. Same on next line

PR #17825 (2016-10-18): Correct the angles in the spectrum viewer
The angles in the spectrum viewer were being calculated incorrectly, this should fix them

**To test:**

1. Start mantidplot
1. Load some data
1. right click ws show detectors
1. right click ws spectrum viewer
1. Compare the TwoTheta, and Azumithal values from the spectrumviewer with the corresponding row in the detectors table using the theta and phi columns

Fixes #16792.

**RELEASE NOTES**
docs/source/release/.../ui.rst

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testin [... cut]
closed 2016-10-18 by AndreiSavici