mantidproject/mantid #16794: ConvertUnits Fails on First Run

EltayebAhmed, 2016-07-05:
### Expected behavior
ConvertUnits Should work first time while doing an inplace unit conversion
### Actual behavior
ConvertUnits fails on first trial.it  Succeeds on Second trial. 

### Steps to reproduce the behavior
Terminal log Containing Behaviour 
File is 1642000.nxs is part of sample training data
[164200.zip](https://github.com/mantidproject/mantid/files/347442/164200.zip)
```
>>>ws3 = Load('164200.nxs')

>>>ws3 = ConvertUnits(ws3,Target = 'DeltaE')
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-82-0c2849963340> in <module>()
----> 1 ws3 = ConvertUnits(ws3,Target = 'DeltaE')

C:\MantidInstall\bin\mantid\simpleapi.pyc in algorithm_wrapper(InputWorkspace,Target,EMode=None,EFixed=None,AlignBins=None, Version=1)
    879 
    880         try:
--> 881             algm.execute()
    882         except RuntimeError, e:
    883             if e.args[0] == 'Some invalid Properties found':

RuntimeError: ConvertUnits: error (see log)

>>>ws3 = ConvertUnits(ws3,Target = 'DeltaE')
>>>
```

### Platforms affected
Tested MantidPlot 3.7.1 on Windows 7

NickDraper, 2016-07-14:
It also sets the unit, and would undo the bin width division of distribution data

NickDraper, 2016-07-14:
## Units that perform validation

#### DeltaE

if (efixed == 0.0)
    throw std::invalid_argument(
        "efixed must be set for energy transfer calculation");

    throw std::invalid_argument(
        "emode must be equal to 1 or 2 for energy transfer calculation");

#### SpinEchoLength

  if (efixed == 0.0)
    throw std::invalid_argument(
        "efixed must be set for spin echo length calculation");
  if (emode > 0) {
    throw std::invalid_argument(
        "emode must be equal to 0 for spin echo length calculation");
  }

PR #16881 (2016-07-21): ConvertUnits could corrupt inplace workspaces on error
Several validation checks were taking place after the output workspace had already been altered.  for In place workspaces this meant the workspace was altered even though the algorithm failed.

The changes delay the creation of the output workspace and changes until all of the validation has passed.

**To test:**

1. All unit tests and system tests must pass
1. Create or load a workspace
1. Try to convert to DeltaE, or SpinEchoLength with an Elastic Emode and not setting eFixed.
1. Check that the input workspace has not changed (check the  [... cut]
closed 2016-07-21 by FedeMPouzols