mantidproject/mantid #16795: Built-in Lorentz Fit Failing

LamarMoore, 2016-07-05:
*This issue is based on feedback I received on the Mantid training course held at the MLZ in Munich June 30 - July 1.*

Below is the email I received from Dr Wiebke Lohstroh

*As discussed last week at the Mantid workshop in Garching, I send you a short summary of my difficulties with the built-in Lorentz function:* 

*I tried to fit my data with a convolution of my measured resolution function with the sum of two Lorentzians, i.e. :*

*I = Res convoluted (Lor1 + Lor2)*

*If I use the built-in Lorentz function, I get strange steps in the calculated data (see attached files, the employed function strings are also enclosed). If I enter the formula for the Lorentz manually, everything works fine (and gives reasonable results).* 

[Lorentz_fit [Compatibility Mode].pdf](https://github.com/mantidproject/mantid/files/347681/Lorentz_fit.Compatibility.Mode.pdf)

SpencerHowells, 2016-07-05:
There are 2 'Lorentzian' functions: Lorentz includes a flat-background; Lorentzian doesn't. When we do a convolution fit we use Lorentzian. If you use Lorentz you will be convolving with the background so goodness knows what that does! We fit a straightforward background plus the res*Lorentzian.

SpencerHowells, 2016-07-05:
Just noticed the pdf & read it. The discontinuity in fit of #1 looks as if it is due to the Fit parameter Peak Radius is too small - using the default value. When using a Lorentzian it needs to be changed to a much larger value using View>Preferences>Mantid>CurveFitting - I use 50. Muons have also seen this problem and Roman is aware of the Issue. 
For Indirect Inelasic (QENS) we have instructions at http://www.mantidproject.org/Indirect:Indirect_Data_Analysis#Running_ConvFit
and a fitting primer at
 http://www.mantidproject.org/Fitting_QENS_Peaks

We have been using this fitting procedure for many years with no problems.

LamarMoore, 2016-07-06:
Thank you for the feedback @SpencerHowells I will pass it on.

closed 2016-10-14 by NickDraper