mantidproject/mantid #16800: Refactor LoadNexusLogs to use PropertyNexus

FedeMPouzols, 2016-07-05:
[LoadNexusLogs::createTimeSeries](https://github.com/mantidproject/mantid/blob/master/Framework/DataHandling/src/LoadNexusLogs.cpp#L504) looks like a duplicate of  [PropertyNexus::loadProperty](https://github.com/mantidproject/mantid/blob/master/Framework/API/src/PropertyNexus.cpp#L112). As we have  `PropertyNexus`  which encapsulates the loading of property values from NeXus files, including time series, `LoadNexusLogs` could just just `PropertyNexus` to do the loading here. There are other places in the algorithm where scalar values could probably be loaded with `PropertyNexus` as well.

See #16345 for a recent refactoring of PropertyNexus.

PranavBahuguna, 2016-07-20:
The two methods don't match exactly here - should I simply replace identical code in LoadNexusLogs::createTimeSeries with the PropertyNexus::loadProperty method and leave other code alone, or should I replace the former entirely with latter altogether?

FedeMPouzols, 2016-07-20:
Where do they differ? The aim here would be to use `PropertyNexus::loadProperty` and not repeat its code. Now if `createTimeSeries` is doing more than that, then it would be perfectly fine to leave the additional code.

PranavBahuguna, 2016-07-20:
Should `PropertyNexus::loadProperty` itself be altered?

FedeMPouzols, 2016-07-20:
That wasn't the original idea, but if there's a good reason why not. Where do you see it should be changed?

PranavBahuguna, 2016-07-22:
On closer analysis of `LoadNexusLogs::createTimeSeries` and `PropertyNexus::loadProperty`, I do not believe a refactor is possible. 

Firstly, the former method performs numerous checks on the attributes and data it loads that is not done in `loadProperty`.

Another issue is that many actions involved in accessing data relevant data in `LoadNexusLogs` are only possible with the specific functions it uses in `createTimeSeries`. For example, time data is obtained with `getDataCoerce`, but attempting to achieve the same thing with loadProperty's `getData` does not work. Thus it would be necessary make significant changes to `loadProperty` in order to replace the functionality of `createTimeSeries`. Since `loadProperty` is used by other classes, I feel it would be unwise to change the method so much just to work with a specific class and data.

closed 2017-01-16 by NickDraper