mantidproject/mantid #16829: Mantid fails to build with CMake 3.6

quantumsteve, 2016-07-09:
### Expected behavior

Mantid builds after upgrading to CMake 3.6 on OS X.

### Actual behavior

CMake fails to find HDF5.

```
-- HDF5: Using hdf5 compiler wrapper to determine CXX configuration
CMake Error at /usr/local/Cellar/cmake/3.6.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:148 (message):
  Could NOT find HDF5 (missing: HDF5_HL_LIBRARIES) (found version "1.8.16")
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.6.0/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:388 (_FPHSA_FAILURE_MESSAGE)
  /usr/local/Cellar/cmake/3.6.0/share/cmake/Modules/FindHDF5.cmake:707 (find_package_handle_standard_args)
  buildconfig/CMake/CommonSetup.cmake:82 (find_package)
  CMakeLists.txt:63 (include)

-- Configuring incomplete, errors occurred!
```

2. Linking erros when building Mantid. For example,

```
[709/709] Linking CXX executable bin/DataHandlingTest
FAILED: bin/DataHandlingTest 
: && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -Wall -Wextra -Wconversion -Winit-self -Wpointer-arith -Wcast-qual -Wcast-align -fno-common -Wno-deprecated -Wno-write-strings -Wno-sign-conversion -Woverloaded-virtual -fno-operator-names -O3 -DNDEBUG -Wl,-search_paths_first -Wl,-headerpad_max_install_names  Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/XMLInstrumentParameterTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/UpdateInstrumentFromFileTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/StartAndEndTimeFromNexusFileExtractorTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SortTableWorkspaceTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SetScalingPSDTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SetSampleTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SetSampleMaterialTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SetBeamTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveToSNSHistogramNexusTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveSavuTomoConfigTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveSPETest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveReflThreeColumnAsciiTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveTBLTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveReflCustomAsciiTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveRKHTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveParameterFileTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SavePHXTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SavePDFGuiTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SavePARTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveNexusTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveNexusProcessedTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveNXcanSASTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveNXTomoTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveNXSPETest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveNISTDATTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveMaskTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveIsawDetCalTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveILLCosmosAsciiTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveGSSTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveGSASInstrumentFileTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveFullprofResolutionTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveFocussedXYETest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveFITSTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveDspacemapTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveDiffCalTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveDetectorsGroupingTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveDaveGrpTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveCanSAS1dTest2.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveCanSAS1dTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveCalFileTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveCSVTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveAsciiTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveAscii2Test.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SaveANSTOAsciiTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/SNSDataArchiveTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/RotateSourceTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/RotateInstrumentComponentTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/RenameLogTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/RemoveLogsTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/RawFileInfoTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/ProcessDasNexusLogTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/PDLoadCharacterizationsTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/NexusTesterTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/MoveInstrumentComponentTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/ModifyDetectorDotDatFileTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/MergeLogsTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/MaskDetectorsTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/MaskDetectorsInShapeTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadVulcanCalFileTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadTOFRawNexusTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadSwansTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadSpiceXML2DDetTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadSpiceAsciiTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadSpice2dTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadSavuTomoConfigTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadSaveAsciiTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadSassenaTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadSPETest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadSNSspecTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadSINQFocusTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadTBLTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadRawSpectrum0Test.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadRawSaveNxsLoadNxsTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadRawBin0Test.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadRaw3Test.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadRKHTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadQKKTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadPreNexusTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadPreNexusMonitorsTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadParameterFileTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadPDFgetNFileTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadNexusTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadNexusProcessedTest.cpp.o Framework/DataHandling/test/CMakeFiles/DataHandlingTest.dir/LoadNexusMonitorsTest.cpp.o Framew [... cut ...]
      SaveNXSPETest::saveAndReloadWorkspace(boost::shared_ptr<Mantid::API::MatrixWorkspace>) in SaveNXSPETest.cpp.o
  "_H5LTread_dataset_double", referenced from:
      SaveNXSPETest::saveAndReloadWorkspace(boost::shared_ptr<Mantid::API::MatrixWorkspace>) in SaveNXSPETest.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

### Steps to reproduce the behavior

Build Mantid with CMake 3.6

### Platforms affected

Any platform with CMake 3.6

quantumsteve, 2016-07-09:
On OS X, FIndHDF5 work only after adding the hint `-DHDF5_ROOT=/usr/local/opt/hdf5`.

quantumsteve, 2016-07-09:
The linking errors go away if I add `${HDF5_HL_LIBRARIES}`

quantumsteve, 2016-07-09:
@NickDraper @peterfpeterson I've worked around the issues on OS X, but one should check for issues on Windows and Linux.

quantumsteve, 2016-07-12:
CMake issue and merge request
https://gitlab.kitware.com/cmake/cmake/issues/16190
https://gitlab.kitware.com/cmake/cmake/merge_requests/34

quantumsteve, 2016-07-14:
I've worked around issues with CMake 3.6 on OS X and @peterfpeterson has built CMake 3.6 on Fedora 23 and verified that it finds HDF5 and builds Mantid. 

@NickDraper Would you install CMake 3.6 on Windows and make sure that it finds the HDF5 libraries and successfully build Mantid?

NickDraper, 2016-07-19:
Configure, Generate and build (MAKE_VATES Off) seems to be fine on Windows 7.

Note:
They have changed the installer on 3.4, so you have to manually uninstall the last version before installing 3.6.

quantumsteve, 2016-07-19:
@VickieLynch and @ricleal have encountered issues with CMake 3.6 on RHEL 7. I _think_ we need to set `HDF5_ROOT` before FindHDF5 so that it actually finds HDF5.

AntonPiccardoSelg, 2016-07-20:
@quantumsteve  Built Mantid(with Vates and PV5.1.0) with Cmake3.6.0 on Windows 7 without any issues.

NickDraper, 2016-10-14:
does not seem to be a current problem

closed 2016-10-14 by NickDraper