mantidproject/mantid #16830: ParaView fails to build with CMake 3.6

quantumsteve, 2016-07-09:
### Expected behavior

### Actual behavior

### Steps to reproduce the behavior

### Platforms affected

quantumsteve, 2016-07-09:
On OS X, FIndHDF5 works only after adding the hint -DHDF5_ROOT=/usr/local/opt/hdf5. 

Cherry-picking [this commit](https://github.com/Kitware/VTK/commit/351d0a6d1ce13daef7e52ed988f843aae0ff9fbf) appears to fix the linking errors.

quantumsteve, 2016-07-09:
I updated our OS X buildscripts to workaround the issues I encountered building v5 and v5.1. 

 https://github.com/mantidproject/paraview-build/commit/e4832ad8d73a8536174108d87090f61b6c248557

https://github.com/mantidproject/paraview-build/commit/c4d9c70734fc2f7f9f8d84e71ddb12dc22457118

quantumsteve, 2016-07-09:
@NickDraper @peterfpeterson This should also be checked on Windows and Linux

quantumsteve, 2016-07-12:
CMake issue and merge request
https://gitlab.kitware.com/cmake/cmake/issues/16190
https://gitlab.kitware.com/cmake/cmake/merge_requests/34

VTK merge request
https://gitlab.kitware.com/vtk/vtk/merge_requests/1679

peterfpeterson, 2016-07-13:
Do we need to wait until kitware moves forward before we suggest cmake 3.6 to developers?

quantumsteve, 2016-07-13:
@peterfpeterson Mantid has the same issues with the new FindHDF5, which we'll have to work around (see #16829). We'd benefit from at least one early adopter on Windows, Linux, and OS X.

quantumsteve, 2016-07-14:
I've worked around the build issues with CMake 3.6 on OS X and verified that the linking errors will be addressed in ParaView 5.1.1.

@NickDraper @peterfpeterson Would you please assign one or more people to verify we can still build ParaView with CMake 3.6 on Windows and Linux?

peterfpeterson, 2016-07-15:
Works just fine on fc23

NickDraper, 2016-07-19:
Anton can you check that we can still build ParaView with CMake 3.6 on Windows?

AntonPiccardoSelg, 2016-07-19:
@quantumsteve 
Using the builder scripts I built PV (5.1.0) with Cmake3.6.0 on Windows 7.

The build compiled without errors but with loads of warnings ( I think they are normally there). PV runs fine.

AntonPiccardoSelg, 2016-07-20:
@quantumsteve  After building PV (5.1.0) with Cmake3.6.0 on Windows 7 with an up-tod-ate third-party-folder, it still appears to be fine.

peterfpeterson, 2017-05-23:
We're using cmake 3.8

closed 2017-05-23 by peterfpeterson