mantidproject/mantid #16832: Do not use `Histogram` to store variances

SimonHeybrock, 2016-07-11:
As described in the [HistogramData documentation](http://docs.mantidproject.org/nightly/concepts/HistogramData.html) and the [design document](https://github.com/mantidproject/documents/blob/master/Design/histogram_type.md), the `E` data in an `ISpectrum` is, in various places, converted to variances (by squaring) and converted back to standard deviations at a later point. This is dangerous and must be avoided.

The purpose of this issue is to refactor code that does this. The following steps are involved:

1. Survey code and find all cases where this is done.
2. Understand & design strategy, there are several options:
  - Keep storing standard deviations, eliminate cases where `E` is used as variances.
  - Always store variances, eliminate cases where `E` is used as standard deviations.
  - Switch on write (just like done for `setBinEdges()` and `setPoints()`), eliminate cases that have a fixed interpretation of `E`, use the `variances()` and `standardDeviations()` interface instead.
3. Refactor.

For 2.) we also have to consider overhead by computing `sqrt`. For example, `rebin()` has a mode that returns variances, such that adding many histograms is cheaper. Regardless of how `Histogram` will store data internally, this could be solved by simply using `CountStandardDeviations` and constructing `Histogram` only once all additions have been done.

@LamarMoore @OwenArnold part from the simple cases of refactoring, I think this is one of the next things we should consider doing with `Histogram` (or find someone to do it).

LamarMoore, 2016-07-11:
`NormaliseToMonitor` and `DiffractionFocussing2` need a `rebin` method in HistogramData to replace the one in `VectorHelper.h`

SimonHeybrock, 2016-07-11:
Regarding @LamarMoore 's comment, note that `VectorHelper::rebinHistogram()` is storing either variances or standard deviations in the out `E` vector.

closed 2021-03-03 by stale[bot]