mantidproject/mantid #16873: Incorrectly rendered latex formulae in VSI with matplotlib < 1.3

martyngigg, 2016-07-14:
### Expected behavior
Equations should be displayed as expected.

### Actual behavior

On RHEL7:
![vsi-latex-rhel7](https://cloud.githubusercontent.com/assets/733773/16832958/cfe2159e-49a4-11e6-9a05-4b2648af0eac.png)

while on Ubuntu with mpl v1.3.1:
![vsi-latex-ubuntu](https://cloud.githubusercontent.com/assets/733773/16832994/03f58eec-49a5-11e6-82b9-f2bfafca1e92.png)

### Steps to reproduce the behavior
This will not produce the above picture but a simplified version that demonstrates the same error.

Run the following script:
```python
CreateSimulationWorkspace(Instrument='mari', BinParams='-90,0.01,100', OutputWorkspace='mari')
AddSampleLog(Workspace='mari', LogName='ei', LogText='90.1', LogType='Number')
ConvertToMD(InputWorkspace='mari', QDimensions='Q3D', OutputWorkspace='mdws')
```
and open the `mdws` in the VSI. On RHEL 7 you notice that the superscript is not drawn correctly. It should be a -1 but instead a symbol is drawn.

### Platforms affected

I think this just affects those platforms with mpl < 1.3: RHEL 6, RHEL 7

quantumsteve, 2016-07-14:
Stuart backported matplotlib 1.3.1 for RHEL 7 and put them on packages.sns.gov. @stuartcampbell Are they on copr or another more accessible location?

quantumsteve, 2016-07-15:
Are there enough RHEL 6 machines left to justify an updated matplotlib rpm? What version does `pip install matplotlib` provide on RHEL 6?

martyngigg, 2016-07-18:
We have only a few instrument cabin PCs that will be around until December. As there is a workaround it would probably be okay to just update the RHEL 7 packages.

martyngigg, 2016-09-01:
@stuartcampbell Would it be possible to build `matplotlib` 1.5.1 packages for RHEL 7?

MSlice is going to need it and it will fix this issue too.

stuartcampbell, 2016-09-11:
@martyngigg I've been trying to get 2.0 working...

martyngigg, 2016-09-12:
@stuartcampbell How's it going?

martyngigg, 2016-10-26:
@stuartcampbell I can see `matplotlib-1.5.2` in [mantid/mantid copr](https://copr.fedorainfracloud.org/coprs/mantid/mantid/package/python-matplotlib/). Are they ready to go or are they the ones that we were having trouble with?

stuartcampbell, 2016-10-26:
@martyngigg To my knowledge they seem to work ok.  It was v2.0 that I was having issues with.

martyngigg, 2016-10-26:
@stuartcampbell Thanks. Are we okay at putting the minimum matplotlib version to 1.5 for RHEL 7 then?

stuartcampbell, 2016-10-26:
@martyngigg I think so, what tests should we do before pulling the 'trigger'?

martyngigg, 2016-10-27:
At minimum we should run the above test with the VSI. I would then also try out some basic `matplotlib` stuff in and outside MantidPlot.

martyngigg, 2016-11-07:
I've just tried this out on a RHEL7 box with the RPMs from copr. It looks good:

![vsi-latex-rhel7-mpl1 5](https://cloud.githubusercontent.com/assets/733773/20064023/1a49b486-a501-11e6-9e59-4f6edee7573b.png)

martyngigg, 2016-11-07:
I have the changes at https://github.com/mantidproject/mantid/compare/master...matplotlib-1.5-el7. I'm going to put the RPMs from copr in the ISIS repo before opening the pull request. Do they need to be in the SNS repo too?

martyngigg, 2016-11-08:
@stuartcampbell Should the dependency be `python2-matplotlib` or `python2-matplotlib`?

PR #18023 (2017-01-16): Require python2-matplotlib >= 1.5.2 on RHEL 7
Places a minimum requirement of matplotlib >= 1.5.2 on the RHEL 7 rpm. The rpms were built on the [mantid copr](https://copr.fedorainfracloud.org/coprs/mantid/mantid/) and are also in the ISIS repository.

**To test:**

Build a package on RHEL 7 and verify it requires `python2-matplotlib >= 1.5.2`

Fixes #16873

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidproject.org/Individual_Ticket_Testing)):

##### Code Review #####

- [ ] Is the code of an acceptable [... cut]
closed 2017-01-16 by SimonHeybrock