mantidproject/mantid #16878: HistogramData Rollout: FrameWork/Algorithms/XDataConverter

LamarMoore, 2016-07-14:
This issue relates to the broader HistogramData rollout see [here](http://docs.mantidproject.org/nightly/concepts/HistogramData.html#rollout-status) and [here](https://github.com/mantidproject/mantid/issues/16872). We are currently refactoring code in order to make use of the new interface by replacing references to the old interface and making a few improvements where possible. In this issue you will be modifying XDataConverter.cpp.

### Examples

Examples of changes include:
 - `MantidVec &X = inputWS->dataX(0);` becomes `auto &X = inputWS->mutableX(0);`
 - `outputWS->dataX(i) = inputWS->readX(i);` becomes `outputWS->setSharedX(i, inputWS->sharedX(i));`
 - 
```c 
outputWS->setX(i, inputWS->refX(i));
outputWS->dataY(i) = inputWS->readY(i);
outputWS->dataE(i) = intputWS->readE(i);
```
becomes
```
outputWS->setHistogram(i, inputWS->histogram(i));
```
### Tasks
There are a few tasks which should be completed when working on these files:
1.Add performance tests where they may be missing. Read [this](http://www.mantidproject.org/Writing_Performance_Tests) document carefully for information on how to create and run performance tests.
2. Run the SystemTest/PerformanceTest for the algorithm before making any changes and make a note of the execution time. How to run unit tests can be found [here](http://www.mantidproject.org/Running_the_unit_tests).
3. Modify XDataConverter.cpp as well as its unit test (XDataConverter.cpp/XDataConverterTest.cpp) to conform to the new Histogram standard.
3. After making changes run the SystemTest/Performance test and make sure the execution time has not increased significantly.

PranavBahuguna, 2016-07-25:
On analysis of XDataConverter algorithm, it does not appear to have any documentation nor any unit tests at all. Could you explain why this is so?

FedeMPouzols, 2016-07-25:
@PranavBahuguna: XDataConverter is [a base class](https://github.com/mantidproject/mantid/blob/master/Framework/Algorithms/inc/MantidAlgorithms/XDataConverter.h#L16). It seems that the only concrete algorithm that uses it is `ConvertToHistogram`.

PranavBahuguna, 2016-07-25:
In `XDataConverter::setXData`, there are functions that require x data in MantidVec form - this is done by the old functions `readX` and `dataX` but not by the new ones (which give it in `HistogramX` form. Should I leave them, or should I attempt to change these as well?

FedeMPouzols, 2016-07-25:
@PranavBahuguna: I'd check with @LamarMoore what's the recommended approach.

PranavBahuguna, 2016-07-26:
@LamarMoore As mentioned in my above comment, what approach should I take? For example from `setXData`, `ConvertToPointData::calculateXPoints` calls `Kernel::VectorHelper::convertToBinCentre`, which requires parameters of form `MantidVec`. I cannot use the new functions as they do not give the results in that form.

PR #17265 (2016-08-24): HistogramData Rollout: XDataConverter
As part of the HistogramData Rollout, the existing code for `XDataConverter` and its unit tests has been refactored to make use of the new methods in HistogramData. In addition, I have also written performance tests to verify that this refactor has not resulted in an increase in execution time.

Fixes #16878 .

**To test:**

Run performance tests for `ConvertToHistogramTest` and `ConvertToPointDataTest` - the execution time for both tests should be reduced from around 1.1s to 0.3s.

Release notes have been updated.

#### Reviewer ####

Please comment on th [... cut]
closed 2016-08-24 by martyngigg