mantidproject/mantid #16966: Indirect BayesQuasi - Refactor addition of Sample logs

ElliotAOram, 2016-07-21:
### Expected behavior
* The BayesQuasi algorithm should use the IndirectBayes.py script as little as possible. If any of the imports it currently uses are only used by BayesQuasi then refactor them into the algorithm. If other algorithms (check ResNorm2.py and [BayesStretch](https://github.com/mantidproject/mantid/pull/16944) ) use these imports we can discuss how it is best to deal with them. 

### Actual behavior
* Currently it imports the following:
```
from IndirectBayes import (CalcErange, GetXYE, ReadNormFile, ReadWidthFile, QLAddSampleLogs, C2Fw, C2Se, QuasiPlot)
```

### Steps to reproduce the behavior
* Code review

### Platforms affected
* All

ElliotAOram, 2016-07-22:
Once you refactor the functions from IndirectBayes -> BayesQuasi make sure you delete the functions from IndirectBayes. The unit tests should then fail for any other algorithms depending on these (although I think it should be none!)

SpencerHowells, 2016-07-25:
This algo reads ascii files in readAsciiFile & ReadWidthFile. Shouldn't we take this opportunity to change 'r' to 'U' in the handle call?

ElliotAOram, 2016-07-25:
Good idea! @louisemccann can you also makes this change?

SpencerHowells, 2016-07-25:
Do we still have Plot & Save in the algo rather than the gui?

ElliotAOram, 2016-07-25:
@SpencerHowells yes I noticed this the other day. I must have forgotten to move it over when I initially rewrote this. I think given the size of the issue it might be worth creating a separate issue

SpencerHowells, 2016-07-25:
I agree.

ElliotAOram, 2016-07-25:
#17012

PR #17003 (2016-08-01): Indirect - Update BayesQuasi Algorithm
The BayesQuasi algorithm uses functions from the IndirectBayes script. The functions used have been moved directly to the BayesQuasi algorithm and removed from the IndirectBayes script. Have also reduced the amount of time the algorithm takes to produce sample logs.

**To test:**

Run BayesQuasi using the example at [http://docs.mantidproject.org/nightly/algorithms/BayesQuasi-v1.html](url)

Fixes #16966 and #16955

*Does not need to be in the release notes.*

---

#### Reviewer ####

Please comment on the following ([full description](http://www.mantidpro [... cut]
closed 2016-08-01 by FedeMPouzols