mantidproject/mantid #16979: Create new unit DeltaE_MicroEV

gvardany, 2016-07-22:
Add a unit for DeltaE in micro electron volts.

stuartcampbell, 2016-07-22:
@jmborr Would this be useful to include in `BASISReduction` when done ?

jmborr, 2016-07-22:
I don't think so because the instrument scientists are getting used to the
Mantid analysis interface, which assumes meV. The only reason they like
micro-eV is because they still use DAVE.
By the way, BASISReduction saves the reduced file in nexus and DAVE
formats, the latter in micro-eV.

SpencerHowells, 2016-07-22:
ISIS users have also got used to using meV.

jmborr, 2016-07-22:
@gvardany What is the need for this unit?

SpencerHowells, 2016-07-23:
I agree. The whole Indirect package works with meV. Another unit or different scale would bring confusion & problems. In the absorption corrections energy is converted to wavelength. There are routines that transform to I(q,t) - the time scale/range will change. The JumpFit interface does model fitting to the widths - again scale/units problems. Finally, in many of the interfaces there are features that work in meV.

gvardany, 2016-07-25:
The main reason was again to be one-to-one compatible with the LAMP software at ILL. We are still discussing this, but it seems this is not a very strict requirement. Probably we will roll back to meV to be consistent within Mantid.

NickDraper, 2016-07-25:
Thanks for considering this @gvardany if we can keep a consistent, reasonable list of units it will control the list getting out of hand.  We are longer term looking at allowing unit transformations to cover scale transformations within similar units, but that is likely to be 2+ years away.

Of course if this becomes a major issue we can revisit it.

gvardany, 2016-07-25:
Put off.

closed 2016-07-25 by gvardany