mantidproject/mantid #16983: Add global options to `Settings` tab

raquelalvarezbanos, 2016-07-22:
This issue is to:

- Gather requirements from scientists about the global options that should be shown in the `Settings` tab.
- Replace the existing `HintingLineEdit`s. This implies:
 - Removing unnecessary includes
 - Using initializer list to initialize the presenter in the constructor of `QtReflSettingsTabView`'s constructor

In addition:
 - Ensure that `ReflSettingsPresenter` knows as little as possible about how the algorithms are run.

raquelalvarezbanos, 2016-07-26:
Tentative list of global options:

- Multidetector/point detector option

- Detectorlimits={'low': 395, 'specular': 404, 'high': 414} 
These give the limits of integration for a multi-detector and must be separately addressable

- binningpars = ["1.0","0.02","14.0"]
Binning parameters for the direct beam (transmission workspace)

- dbnr = 'none'
The direct beam (transmission workspace)

- savepath = "u:/randomfiles/"
savepath

- nrqlimits = ["0.008","-0.02","0.2"]
Final binning parameters for the stitched data

- nrscalefactor =1.0
scale

- nrqgroup=False
should the detector be summed in lambda or in q?

- nrbackground=False
Should a background subtraction happen?

- ofqlimits = '-5e-4,5e-4,0.02,0.1'
ofNqx = 500
ofNqz = 500
offspecular limits and numbers of points

- lmin, lmax = 1.5 , 12.0
the minimum and maximum lambda to use to generate the cutoffs in Q

- calibration = nr.PolarisationCalibration.current
Set the correct polarisation correction, this will be the values for alpha, rho pp and ap

raquelalvarezbanos, 2016-07-26:
Jos sent this example class they use to set different types of polarisation corrections:

```
class PolarisationCalibration():
    calibrations = WeakValueDictionary()
    
    def __init__(self, name, rhocoefficients, alphacoefficients, Apcoefficients, Ppcoefficients, runs = None, comment = None):
        self.rhocoefficients = rhocoefficients
        self.alphacoefficients = alphacoefficients
        self.Apcoefficients = Apcoefficients
        self.Ppcoefficients = Ppcoefficients
        self.name = name
        self.comment = comment
        PolarisationCalibration.calibrations[name] = self
        self.current = False
        
    @classmethod
    def get_calibrations(cls):
        return cls.calibrations
    @classmethod
    def set_current_calibration(cls, instancename):
        thecalibration = cls.calibrations[instancename]
        thecalibration.current = True
        cls.current = thecalibration
        return thecalibration
  

   
nocorr = PolarisationCalibration('nocorr',  rhocoefficients=[1.0],   alphacoefficients=[1.0], Apcoefficients=[1.0], Ppcoefficients=[1.0], runs = [00000], comment = "all are = 1.0")

    
oldest = PolarisationCalibration('oldest',rhocoefficients=[1.006831,-0.011467,0.002244,-0.000095], alphacoefficients=[1.017526,-0.017183,0.003136,-0.000140], Apcoefficients=[0.917940,0.038265,-0.006645,0.000282], Ppcoefficients=[0.972762,0.001828,-0.000261,0.0], runs = [1850,1855,1851,1856], comment = "Constants Based on Runs 18350+18355 and 18351+18356 analyser theta at -0.1deg 2 RF Flippers as the polarising system")   
   
december2012 = PolarisationCalibration('december2012',  rhocoefficients=[0.970257,0.016127,-0.002318,0.000090],   alphacoefficients=[0.975722,0.012464,-0.002408,0.000105], Apcoefficients=[1.030894,-0.040847,0.006069,-0.000247], Ppcoefficients=[0.961900,-0.003722,0.001094,-0.000057], runs = [i for i in range(19438,19460)], comment = "Drabkin on incident side RF Flipper on analyser side Dec 2012")

march2016 = PolarisationCalibration('march2016', rhocoefficients=[1.00002594,-0.004658,-0.000336,0.000023], alphacoefficients=[1.01626,-0.006871,0.000509,-0.00002], Apcoefficients=[0.971155,-0.00469,0.000601,-0.000071], Ppcoefficients=[1.133720,-0.152533,0.050698,-0.008468,0.000699,-0.000022], runs=[i for i in range(37660,37709)], comment= "RF system removed, Drabkin incident between MagA and B and double Vs between C and D, analyser at nominal -0.1")

february2013_2 = PolarisationCalibration('february2013_2', rhocoefficients=[0.945927,0.025421,-0.003647,0.000156], alphacoefficients=[0.940769,0.027250,-0.003848,0.000164],  Apcoefficients=[0.974374,-0.005334,0.001313,-0.000115], Ppcoefficients=[1.023141,-0.024548,0.003398,-0.000134], runs =  [i for i in range(19628,19671)], comment = "analyser -0.2deg RF Flippers on polariser and analyser side Feb 2013")

february2013_1 = PolarisationCalibration('february2013_1', rhocoefficients=[0.955384,0.021501,-0.002962,0.000112], alphacoefficients=[0.944022,0.025639,-0.0035,0.000145],  Apcoefficients=[0.980605,-0.002201,-0.00097,0.000014], Ppcoefficients=[0.992212,-0.009161,0.000601,0.000025], runs =  [i for i in range(19628,19671)], comment = "analyser -0.1deg RF Flippers on polariser and analyser side Feb 2013")

november2015 = PolarisationCalibration('november2015', rhocoefficients=[0.944514,0.026548,-0.003837,0.000168], alphacoefficients=[0.944022,0.025639,-0.0035,0.000145], Apcoefficients=[0.980605,-0.002201,-0.00097,0.000014], Ppcoefficients=[0.992212,-0.009161,0.000601,0.000025], runs =  [i for i in range(36680,36695)], comment = "analyser -0.1deg RF Flippers on polariser and analyser side 0.5MHz November 2015")

november2015_2 = PolarisationCalibration('november2015_2', rhocoefficients=[0.702221,0.229436,-0.067308,0.009353,-0.000619,0.000016], alphacoefficients=[0.697757,0.230859,-0.067212,0.009276,-0.000610,0.000015], Apcoefficients=[1.065455,-0.067522,0.017810,-0.002430,0.000147,-3.298886e-6], Ppcoefficients=[1.030972,-0.046076,0.014934,-0.002600,0.000226,-7.240769e-06], runs =  [i for i in range(36680,36695)], comment = "analyser -0.1deg RF Flippers on polariser and analyser side 0.5MHz November 2015")

february2016 = PolarisationCalibration('february2016', rhocoefficients=[0.998515,-0.001878,-0.000902,0.000057], alphacoefficients=[1.020642,-0.008133,0.000511,-0.000008346674], Apcoefficients=[0.959164,0.000919,-0.000169,-0.00004], Ppcoefficients=[-0.470963,2.055378,-1.193303,0.363635,-0.0063189,0.006286,-0.000333], runs=[i for i in range(37660,37709)], comment= "RF system removed, Drabkin incident between MagA and B and double Vs between C and D, analyser at nominal -0.1")

alongbeamspecial2016 = PolarisationCalibration('alongbeamspecial2016', rhocoefficients=[1.00002594,-0.004658,-0.000336,0.000023], alphacoefficients=[0.999856,0.001013,-0.000645,0.000017], Apcoefficients=[-0.276464,1.16719,-0.440508,0.084601,-0.008725,0.000458153,-0.000009597066], Ppcoefficients=[1.133720,-0.152533,0.050698,-0.008468,0.000699,-0.000022], runs=[i for i in range(37660,37709)], comment= "RF system removed, Drabkin incident between MagA and B and double Vs between C and D, analyser at nominal -0.1. Guide field along beam at sample position. This measurement is done at max field ~110mT. Kept incident coefficients from march2016 (better stats) and adjusted scattered beam side.")

# 7th order fit [1.510208,0.653148,-0.374399,0.110249,-0.018274,0.001717,-0.000085,0.000001703707

PolarisationCalibration.set_current_calibration('march2016')
```

The idea is that they have a few sets of parameters and they may need use one or another. They may also need to start with a specific set for the first `X` runs and then change it for the next runs. This will be achieved with the different groups in Max's design.

raquelalvarezbanos, 2016-10-14:
I assigning this to @PranavBahuguna. He may split this into separate tasks.

raquelalvarezbanos, 2016-10-14:
This is something we definitely want to get done for release 3.9 as scientists have asked for it.

raquelalvarezbanos, 2016-10-14:
@PranavBahuguna I have added some commits so you can have some examples of how to add new widgets to the tab. I would recomend adding new items one by one. For each new item:

- Add the widget to `ReflSettingsTabWidget.ui`. You may find useful using `QtCreator` but this file can also be edited manually.
- Add the necessary functionality. This means updating the following classes: `ReflSettingsTabPresenter.cpp`, `IReflSettingsTabView.h`, `ReflSettingsTabView.h` and `ReflSettingsTabView.cpp`.
- Update the unit tests: you'll need to update `ReflMockObjects.h` and `ReflSettingsTabPresenterTest.h`.

When you have something similar to the figure above (excluding transmission runs and the "Get defaults" button), let me know.

Let me know if you have any questions and I'll be happy to help :)

PranavBahuguna, 2016-10-18:
Regarding the figure above, what do all the entries in the 'Experiment Settings' section titled 'other' actually refer to?

raquelalvarezbanos, 2016-10-19:
@PranavBahuguna: they are still to be defined. What you can do next is have a look at the list above, which was reported by Jos, and check what of the options are actually related to the reduction algorithm `ReflectometryReductionOneAuto`. Those should be added as well.

PranavBahuguna, 2016-10-19:
I have some more questions regarding some of the global options:
- What is `dbnr` supposed to be exactly? A transmission workspace is created when the algorithm is executed (using transmissionRuns 1 and 2), but isn't itself externally supplied to the algorithm as far as I can tell.
- `Savepath` - what are we saving here?
- `nqgroup` - I may have simply forgotten, but where in the algorithm is this decision to sum detectors specifically in lambda or q occur?
- `ofqlimits`, `ofNqx` and `ofNqz` - what are these? There is no mention of them in the reflectometry algorithms.

Furthermore, for all global options, which would belong in "Experiment Settings" and which in "Instrument Settings"?

raquelalvarezbanos, 2016-10-19:
Some of the options above do not correspond to properties of `ReflectometryReductionOneAuto`, for instance `ofqlimits`, `ofNqx`, `ofNqz`. Those refer (I think) to the next step in the analysis, which is `ConvertToReflectometryQ`, so we can forget about those. The same happens with `Savepath`, this will be added in this issue: https://github.com/mantidproject/mantid/issues/17816. `dbnr` refers to direct beam normalization (so it should be added), and `nqgroup` is not implemented yet so we can forget about this one too for the moment.

PranavBahuguna, 2016-10-20:
One more question - the documentation for `ReflectometryReductionOne` also mentions that background subtraction is performed in `ReflectometryReductionOneAuto`. However I cannot find where this is done in the algorithm, if at all. Do you know if and where this is done as the `nrbackground` option concerns it?

raquelalvarezbanos, 2016-10-21:
I have had a look and I cannot find where the background subtraction is performed in `ReflectometryReductionOneAuto`, and as far as I know background subtraction is not implemented yet. It seems that that sentence in the documentation for `ReflectometryReductionOne` is not correct. Therefore, `nrbackground` should not be added to the settings tab for the moment.

Regarding the second issue, I can explain in more detail when I am back, as it is a bit complicated to explain in a comment here, but essentially, that is done in the methods:

```
ReflSettingsTabPresenter::getReductionOptions()
```

All you have to do is follow my examples above, and the algorithm's properties will be correctly populated. Please add properties step by step as I did in the examples, and don't forget to update the unit tests.

raquelalvarezbanos, 2016-10-21:
@PranavBahuguna: Having a look at your last commit, I am not sure `TransmissionLambdaMin` and `TransmissionLambdaMax` need to be added at this stage. I indicated in the description of this issue that transmission runs should not be considered for the moment, maybe it was not clear, but that applies to properties related to transmission runs as well.

I suggest following the steps in my comment 7 days ago to add a new property at a time. Start with the properties you are sure about (for instance it is clear that `MonitorIntegralMin` refers to `MonitorIntegrationWavelengthMin`, `MonitorBackgroundMin` to `MonitorBackgroundWavelengthMin`, etc). Those you are not sure about, leave them for next week when I am back. I think it is better if we add new properties step by step, instead of adding everything at once, so please, add a commit for each property and, of course, if you have any questions, ask :)

closed 2016-11-07 by martyngigg